linorobot/linorobot2 #2: URDF Stage of Install: Joint_state_publisher waiting for robot_description

philmurp, 2021-10-05:
I am getting stuck at the "Visualize the newly created URDF" portion. I am able to edit my description file and colcon build, but then when I launch via robot computer I get stuck at attached screen.

Further steps like launching robot and conctrolling via keyboard works, but URDF and SLAM all get stuck waiting for this robot_description topic.

"joint_state_publisher: Waiting for robot_description to be published on the robot_description topic"

Any ideas on a resolution, been looking up for a while and reinstalled joint state publisher with no luck. My RVIZ on host computer comes up but is empty and I see the robot_description as an unvisualizable topic.

Thanks
[image]

grassjelly, 2021-10-05:
Hey Phil,

You're only launching the URDF's publisher from the robot computer. You'll also need to run the visualizer from the host computer https://github.com/linorobot/linorobot2#22-visualize-the-robot-from-the-host-machine . I updated the README for clearer instructions.

philmurp, 2021-10-05:
Hello,

I have been entering below code in both computers:
  - I entered this code into robot computer: ros2 launch
linorobot2_description description.launch.py
  - And then this code into host computer: ros2 launch linorobot2_viz
robot_model.launch.py

This pulls up the RVIZ GUI on my host computer, but it doesnt show anything
besides axes with no robot in them and lists "robot_description as an
unvisualizable topic". Robot computer continues to show waiting for
publisher of description, not an error but just stays stuck there forever.

It seems like the joint_state_publisher from robot computer isnt sending
anything out or receiving back from host computer. Is there something
within the RVIZ on host computer I can edit to try and connect? Both on
same wifi network if that matters.

grassjelly, 2021-10-05:
You can install the package without using the installer  script on your development computer:
https://github.com/linorobot/linorobot2#2-host-machine--development-computer---gazebo-simulation-optional

and run description.launch.py with the rviz arg set to true https://github.com/linorobot/linorobot2#21-publish-the-urdf-from-the-robot-computer

grassjelly, 2021-10-05:
btw, have you tried changing 'Durability Policy' In RobotModel's config in rviz?
[image]

grassjelly, 2021-10-06:
I can't see the images you uploaded. Could you update your comment please? Thanks.
Have you also tried visualizing the URDF using the default file?

philmurp, 2021-10-06:
Ok my updated file code below. I just changed was the numbers slightly so
nothing at all about the structure of the file:

<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
<xacro:property name="base_length" value="0.3619" />
<xacro:property name="base_width" value="0.2222" />
<xacro:property name="base_height" value="0.0508" />
<xacro:property name="base_mass" value="10" />
<xacro:property name="wheel_radius" value="0.0508" />
<xacro:property name="wheel_width" value="0.0508" />
<xacro:property name="wheel_pos_x" value="0.1524" />
<xacro:property name="wheel_pos_y" value="0.1524" />
<xacro:property name="wheel_pos_z" value="-0.04788" />
<xacro:property name="wheel_mass" value=".6" />
<xacro:property name="wheel_torque" value="20" />
<xacro:property name="laser_pose">
<origin xyz="0.1524 0 0.1905" rpy="0 0 0"/>
</xacro:property>
<xacro:property name="depth_sensor_pose">
<origin xyz="0.2375 0 0.06" rpy="0 0 0"/>
</xacro:property>
</robot>

One other thing I noticed in the launch file, under joint_state_publisher
there is no 'robot_description' it is only under the robot_state_publisher
Node. Could this be why the robot_state_publisher tasks run but then
joint_state_publisher gets stuck and is waiting for robot_description?
Should I try copying robot_descritpion line into the joint_state_publisher
Node?

Node(
package='joint_state_publisher',
executable='joint_state_publisher',
name='joint_state_publisher',
condition=IfCondition(LaunchConfiguration("publish_joints")),
parameters=[
{'use_sim_time': LaunchConfiguration('use_sim_time')}
]
),
Node(
package='robot_state_publisher',
executable='robot_state_publisher',
name='robot_state_publisher',
output='screen',
parameters=[
{
'use_sim_time': LaunchConfiguration('use_sim_time'),
'robot_description': Command(['xacro ', LaunchConfiguration('urdf')])
}
]
),

grassjelly, 2021-10-06:
I can't replicate your error. Loads fine on me tho
[image]

Are you getting anythin on joint_states topic?
`ros2 topic echo joint_states
`

grassjelly, 2021-10-06:
Also could you try a simple pub-sub test just to smoke test that your ros2 network is fine:

On a new terminal:
`ros2 topic pub -r 30 cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"`

On a another terminal:
`ros2 topic echo cmd_vel`

philmurp, 2021-10-06:
In your picture are both windows on robot computer or first one on robot
and second one on the host?

If I try and replicate your code in my host computer for the second window
in your picture:
1) cd dev_ws/src/linorobot2_viz
2) ros2 launch linorobot2_viz robot model.launch.py

Results is error "file 'robot' was not found in the share directory of
package 'linorobot2_viz' which is at
'/home/phil/dev_ws/install/linorobot2_viz/share/linorobot2_viz'

Is this maybe a directly/install issue that my host computer cant find
these files? For example after I installed from section 2.1 in the README,
I didnt have a linorobot2_ws so I manually created a dev_ws using ROS
online tutorial. Perhaps the host computer install didnt work properly? I
tried it a few time and the viz only install and they all ended up the same
way. My host computer I am using VMWARE virtual desktop.

grassjelly, 2021-10-06:
You have a typo on launching robot_model.luanch.py . There should be an underscore between robot and model, hence it looked for the 'robot' file. Did you try the simple pub-sub test?

philmurp, 2021-10-06:
> Everything seems to be working fine. Except the rviz visualization of the
> mesh. Could you post a photo of how rivz looks like when you launch the viz
> file?

Let me know if this picture comes through.

[image: image.png]

grassjelly, 2021-10-06:
No photo uploaded it’s just text 

[image: image.png]

philmurp, 2021-10-06:
> Trying attachment route.
>
> On Wed, Oct 6, 2021 at 10:25 AM Juan Miguel Jimeno <

Also to elaborate on point I made from a while back; once I try to do this
RVIZ, even when I close out and restart the computer the Teensy does this
triple blinking activity and I cant successfully connect to control the
robot even via "ros2 run teleop_twist_keyboard teleop_twist_keyboard".

The code runs fine on command prompt and I get prompts on how to use the
keyboard to control it, but the teensy doesnt respond. So I then have to go
back to Hardware README and run configuration and sampling code. Then come
back and run "ros2 run teleop_twist_keyboard teleop_twist_keyboard" and it
works.

So this RVIZ is doing something that is blocking out the Teensy from
listening to any other commands, no idea why but I cant get it to listen to
anything after RVIZ is attempted.

-Phil

grassjelly, 2021-10-06:
I think it's the rviz that's causing the problem. The robot computer's publishing the joint_states and you can do pub-sub on the host computer too.

Can I confirm that this is the current setup on your host machine (running on VMWARE):
First terminal:

`ros2 launch linorobot2_description description.launch.py
`

Second terminal:
`ros2 launch linorobot2_viz robot_model.launch.py
`

When you run robot_model.launch.py, are you getting any errors from rviz?

philmurp, 2021-10-06:
I agree I think something on my host isnt installed properly.

I sometime get the below:
1) cd dev_ws/src
2) ros2 launch linorobot2_viz robot_model.launch.py

Results: malformed launch argument 'robot_model.launch.py', expected format
'<name>:=<value>'

Then I usually rebuild and it works to pull up the RVIZ.

Perhaps I should reimage and reinstall everything from scratch? I have
always done the full install to enable gazebo, I can try just the RVIZ this
time.

-Phil

grassjelly, 2021-10-06:
That might be worth the try. Would love to hear the results. Closing this for now. Feel free to re-open this again

closed 2021-10-06 by grassjelly

thanhtruongphan, 2024-11-01:
> Also to elaborate on point I made from a while back; once I try to do this RVIZ, even when I close out and restart the computer the Teensy does this triple blinking activity and I cant successfully [...]

hey @philmurp , can I ask you about the ~hardware README that you have refer to... 

I have the same problem like you and I still don't know how to solve... 🥇

hippo5329, 2024-11-01:
You should run bringup in one terminal and run nav/slam in another terminal.

Please check the wiki, https://github.com/hippo5329/linorobot2_hardware/wiki#bring-up-the-robot

hippo5329, 2024-11-01:
"I can run two motor with teleop twist keyboard on a new terminal (of robot computer). But I can't do it on the host machine. Do I need to ```wifi transfer like [this](https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-wifi-transport-and-ld19-lidar-udp-server-without-a-robot-computer) ?"

No. You should not use micro-ros wifi transport. The current serial transport is fine.

hippo5329, 2024-11-01:
For the discovery problem between robot computer (rpi4) and host computer (ubuntu vm), you can try demo publisher/subscriber between them, before running linorobot2. And check topic list and echo. Network broadcasting maybe the cause. I would suggest Jazzy on ubuntu 24.04 which runs micro-ros rmw v2. Or run ubuntu on bare metal host pc. The issue might be resolved.

hippo5329, 2024-11-01:
Which lidar do you use? It seems to be a higher rated lidar. If you are new to ros2/micro-ros, you should start with lower/smaller basic parts to build robot.

You could use rpi5 instead of rpi4. rpi5 is much faster.

thanhtruongphan, 2024-11-02:
> Or run ubuntu on bare metal host pc. The issue might be resolved.

yeah, I know ubuntu on bare metal host pc, dual boot Ubuntu and Windows 11 in my case. I used it two year ago. 
I had complete delete it when doesn't use. 

And now have a problem with **```shrink```**

[image]

I cannot shrink my C-drive, I searched on internet and try to do but still doesnt solve this problem... :(

thanhtruongphan, 2024-11-02:
> Which lidar do you use? It seems to be a higher rated lidar. If you are new to ros2/micro-ros, you should start with lower/smaller basic parts to build robot.
> 
> You could use rpi5 instead of rpi4. rpi5 is much faster.

yeah, I use lidar A1M8

hippo5329, 2024-11-02:
It might be VM delay issue.

If you have USB 3.x external HDD/SSD drive, you can install ubuntu on it and boot from external drive.

Or you may try linorobot2/docker/docker-compose.yaml, there are webtop and kasmvnc. I didn't use them though.

hippo5329, 2024-11-02:
After you run bringup, do not run nav/slam. Check the published messages hz rate from IMU, odom and lidar.

thanhtruongphan, 2024-11-02:
thank you, @hippo5329 :1st_place_medal: 

my Teensy board just broke this morning 🥹 I was still using it yesterday but this morning, when i plugged in the USB power source, the laptop didn't recognize the USB port, i suddenly realized that the power light wasn't on anymore. maybe it's broken. i don't know, why did such an expensive board break, while Arduino and STM32 were fine. hah first time i had a circuit break like that.

hippo5329, 2024-11-02:
I would suggest esp32. It is cheap and powerful. I used a lot of them.

hippo5329, 2024-11-02:
You may try another cable or USB port.

thanhtruongphan, 2024-11-02:
yeah, I have tried, and also try with my laptop, and simple upload code for test, and nothing happen... :D so sad 🥹

hippo5329, 2024-11-02:
You should use "remote development" on your vscode. Teensy is on your robot with rpi4/5. No need to plug into your laptop to program teensy. Check my wiki.

vscode on windows laptop -- (wifi) -- ubuntu on rpi4/5 -- (usb)-- micro-ros on teensy, esp32,pico

hippo5329, 2024-11-02:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#development-on-windows-or-linux-desktop-and-use-vscode-remote-development-ssh-to-rpi45

thanhtruongphan, 2024-11-02:
yeah I known that. I mean, when I think It's broken, I plugin it into laptop by the way most simple to test, but my Teensy still doesn't work, and doesn't show up in the Device manager (COM port)... :face_holding_back_tears:

thanhtruongphan, 2024-11-02:
well, forget it for now. maybe my L298 board doesn't have an isolated op-to, when driving the motor at high speed it causes a short circuit in Teensy?

Last night, when using Teensy to operate linorobot2, I also found it quite hot. I thought it was normal so I didn't care much. but today it broke. I still haven't verified the reason why, I don't want to do it again with Teensy for now. maybe I'll be interested in ESP32.

Oh, and what about STM32. Some popular modules like STM32F4 have 64-256KB of SRAM, 256-512KB of FLASH, about 100MHz freq, are they feasible when implemented with linorobot2? for example this module https://docs.platformio.org/en/latest/boards/ststm32/genericSTM32F411CE.html

actually where I live, Teensy is quite rare and hard to get. Maybe because it breaks easily? :smile:

thanhtruongphan, 2024-11-02:
btw, I have tried a simple publish/subscribe on ROS2, 

>>>   [image]

The listener doesn't listen anything from talker. 
I tried to use ROS_DOMAIN_ID but same as above. :walking: 

i mod ROS_DOMAIN_ID on both two ubuntu
>>>   [image]

hippo5329, 2024-11-02:
Though esp32 is cheap, it is faster than most stm32 and gets more robot related peripherals than stm32, namely the pcnt and pwm. I use eps32 and pico myself. No teensy, no stm32. Expansive parts won't always give you better result.

The ros2 communication issue between hosts is called "discovery". First, you have to make sure they are in the same network. Check the ip, netmask and broadcast. However, I still suggest bare-metal ubuntu on your laptop. It can save a lot of troubles. I think time is more valuable to you. I run windows with KVM on ubuntu when I want to resolve windows issues like yours. I guess most of your data is backuped on ms onedrive. You may restore your data from onedrive.

hippo5329, 2024-11-02:
No need to use ros domain. You should install ubuntu 24.04 as host os and ros2 Jazzy on your laptop.

thanhtruongphan, 2024-11-03:
Yeah, I want to say that I have installed Ubuntu dual boot with Windows11. Is it okay? 

This comment https://github.com/linorobot/linorobot2/issues/2#issuecomment-2453047357 is implemented by this Ubuntu Dual boot 😄 but it's still doesn't work together

hippo5329, 2024-11-03:
Dual boot is fine. Please boot into ubuntu and check the ip of both laptop and rpi4. Try ping each other.

thanhtruongphan, 2024-11-03:
yeah, here is the result... 

[image]

hippo5329, 2024-11-03:
Ping is good.

Try turtlesim on host, check topic echo and hz on rpi. Then turtlesim on rpi, check topic echo and hz on host.

hippo5329, 2024-11-03:
I would suggest install or upgrade to ubuntu 24.04 and Jazzy. There are a lot of improvements.

hippo5329, 2024-11-03:
Also try teleop on either side.

thanhtruongphan, 2024-11-03:
so strange....

[image]

strange, turtlesim should move when i press those keys right? out of those keys only "q" key works :smiling_face_with_tear:

thanhtruongphan, 2024-11-03:
I also tried with the command: 

```
ros2 topic pub /turtle1/cmd_vel geometry_msgs/msg/Twist "linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8"
```
in https://medium.com/@anjitagargi/getting-start-with-ros2-turtlesim-bcb328ff804e
but it's still doesn't work

[image]

hippo5329, 2024-11-03:
https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html

This can be a test to verify the rmw. It should work on the same computer.

Disable the wifi and connect them with ethernet cables to a switching hub. Check if it work with wired connection.

Then try another wifi router/AP. Some wifi router cannot support multicast.

thanhtruongphan, 2024-11-03:
Oh...
Your mean: I should connect Ubuntu laptop with wifi hub via Ethernet cables, to test turtlesim on Ubuntu laptop? 

I have tested with it https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html before, on Ubuntu laptop, it doesn't work. Then I test on SBC computer, it doesn't, too. Then I tested with that, 
```
ros2 topic pub /turtle1/cmd_vel geometry_msgs/msg/Twist "linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8"
```
Its same. Huh

hippo5329, 2024-11-03:
1. on laptop only
   ros2 run turtlesim turtlesim_node # in one terminal
   ros2 run turtlesim turtle_teleop_key # in another terminal

Check if this work?

    ros2 topic list
    ros2 topic hz xxx

2. on rpi4 only

Do the same as 1.

3. on laptop
   ros2 run turtlesim turtlesim_node

on rpi4
   ros2 run turtlesim turtle_teleop_key

hippo5329, 2024-11-03:
ubuntu@cc187:~$ ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
ubuntu@cc187:~$ ros2 topic hz /turtle1/pose
average rate: 62.490
	min: 0.015s max: 0.017s std dev: 0.00050s window: 64
average rate: 62.497
	min: 0.015s max: 0.017s std dev: 0.00050s window: 127
average rate: 62.489
	min: 0.015s max: 0.017s std dev: 0.00050s window: 190
average rate: 62.491
	min: 0.015s max: 0.017s std dev: 0.00050s window: 253

hippo5329, 2024-11-03:
laptop -- wifi -- hub -- wifi--rpi4 # test the wifi connection

laptop--cable--hub--wifi-rpi4

laptop --cable--hub--cable-rpi4  # test the wired connection

thanhtruongphan, 2024-11-03:
Yeah, 

Luckily, I restart all and now 
https://github.com/linorobot/linorobot2/issues/2#issuecomment-2453282556

The 1.  and 2. works.👏
The 3. still no. 

Maybe I need to tried with Ethernet cables. 🫂

thanhtruongphan, 2024-11-03:
It's been 6 years since I've had a laptop, I've never had to use the LAN Ethernet port, it must be rusty now. haha 😄 

now I'm going to buy an Ethernet cable to connect my Ubuntu laptop to Raspi4 to check it out.

Hope it works! 🙏

hippo5329, 2024-11-03:
You will need at least two cables. Both of them need to connected to the hub. Also try a different wifi router.

thanhtruongphan, 2024-11-03:
Oh no? If so, how can I operate the robot car remotely? 🥲

hippo5329, 2024-11-03:
This is for testing only. You do not need wire normally. Be sure to remove ros domain id and restart.

Also try to disable ipv6.

sudo su -
cat <<EOF > /etc/sysctl.d/99-zzz-disable-ipv6.conf
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
EOF
systemctl restart procps

hippo5329, 2024-11-03:
You may ignore the ipv6 thing. The NetworkManager will still enable ipv6.

thanhtruongphan, 2024-11-03:
Oh, i should test it. 

And does it have problem with Wifi 5G ?

hippo5329, 2024-11-03:
The range of wifi 5g is shorter than wifi 2.4g.

I just installed a fresh ubuntu 24.04 and Jazzy on a rpi5 wifi 2.4G. It works with my laptop wifi 5G without issue.

I really think you should use Jazzy instead of humble. The RMW is improved. The cables test may be not needed.

thanhtruongphan, 2024-11-03:
I've bought it https://www.thegioiic.com/day-cap-mang-ethernet-cat5e-2-dau-rj45-duc-dai-10m-mau-vang

hippo5329, 2024-11-03:
I used this cable tests in the early stage of my ros2/micro-ros debugging. I had the discovery issue with humble before.

After I upgrade to jazzy, I don't have such issue.

thanhtruongphan, 2024-11-03:
because I also doing with another project for study that required Ubuntu 22.04, ROS humble, I'm new to ROS so I think I need to do following it before I could modify a little for myself. My project about AMR is using Raspi/Jeston nano + MCU like STM32/arduino to control motor. I also an require to add a 3D camera to track objects (using image processing) but I'm afraid it won't be in time. :smiling_face_with_tear: 

as you refer to, It very helpful for me to do with esp32 and Ros Jazzy, I'll do it soon, for study. currently doing the robot with esp32 is not very suitable for my project.

btw, is this esp32 ok? https://hshop.vn/kit-phat-trien-wifi-ble-soc-esp32-s3-weact-esp32-s3-b-n16r8-espressif-devkitc-1-compatible

[3 more replies]