linorobot/linorobot2 #7: getting error in ekf

raidhu, 2021-11-14:
I'm not using depth sensor.

`[ekf_node-2] Warning: Invalid frame ID "base_footprint" passed to canTransform argument target_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "base_footprint" passed to canTransform argument target_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu_link" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu_link" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu_link" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu_link" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.12/src/buffer_core.cpp
`

grassjelly, 2021-11-14:
I need more information about your system. Could you tell me what's your setup like? Robot base, imu, microcontroller, operating system, ROS distro?

raidhu, 2021-11-14:
robot base : 4wd
imu : 9250
microcontroller : raspberry pi4
operating system : ubuntu20.04.3 (Focal Fossa)
ros : ros2 foxy fitzroy

and I'm not using depth sensor I didn't mention depth sensor while installation is it need to edit workspace or urdf file.

grassjelly, 2021-11-14:
are you getting anything when you echo the following topics?

/imu/data
/odom/unfiltered

raidhu, 2021-11-14:
yes

grassjelly, 2021-11-14:
which launch file does this happen?

raidhu, 2021-11-14:
bringup file

grassjelly, 2021-11-14:
can you post your rqt_graph here?

raidhu, 2021-11-14:
sorry man I can't able to get rqt_graph
it doesn't work properly in ros2

raidhu, 2021-11-14:
everything works fine except efk
 also able to move bot using teleop keyboard.
error I'm facing is

`ubuntu@uvbot:~/linorobot2_ws$ ros2 launch linorobot2_bringup bringup.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2021-11-14-08-47-00-703236-uvbot-3828
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ydlidar_ros2_driver_node-5]: process started with pid [3857]
[INFO] [micro_ros_agent-1]: process started with pid [3849]
[INFO] [ekf_node-2]: process started with pid [3851]
[INFO] [joint_state_publisher-3]: process started with pid [3853]
[INFO] [robot_state_publisher-4]: process started with pid [3855]
[micro_ros_agent-1] [1636879622.171197] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[micro_ros_agent-1] [1636879622.172139] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[micro_ros_agent-1] [1636879622.186130] info     | Root.cpp           | create_client            | create                 | client_key: 0x7C881CDA, session_id: 0x81
[micro_ros_agent-1] [1636879622.186261] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x7C881CDA, address: 0
[micro_ros_agent-1] [1636879622.266785] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x7C881CDA, participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.267979] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x7C881CDA, topic_id: 0x000(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.270327] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x7C881CDA, publisher_id: 0x000(3), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.276638] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x7C881CDA, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[micro_ros_agent-1] [1636879622.278927] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x7C881CDA, topic_id: 0x001(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.281726] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x7C881CDA, publisher_id: 0x001(3), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.284666] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x7C881CDA, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[micro_ros_agent-1] [1636879622.287036] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x7C881CDA, topic_id: 0x002(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.288837] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x7C881CDA, subscriber_id: 0x000(4), participant_id: 0x000(1)
[micro_ros_agent-1] [1636879622.296376] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x7C881CDA, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[ydlidar_ros2_driver_node-5] [INFO] [1636879622.330339628] [ydlidar_ros2_driver_node]: [YDLIDAR INFO] Current ROS Driver Version: 1.0.1
[ydlidar_ros2_driver_node-5]
[ydlidar_ros2_driver_node-5] YDLidar SDK initializing
[ydlidar_ros2_driver_node-5] YDLidar SDK has been initialized
[ydlidar_ros2_driver_node-5] [YDLIDAR]:SDK Version: 1.0.3
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base_link had 7 children
[robot_state_publisher-4] Link camera_link had 1 children
[robot_state_publisher-4] Link camera_depth_link had 0 children
[robot_state_publisher-4] Link front_left_wheel_link had 0 children
[robot_state_publisher-4] Link front_right_wheel_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link laser had 0 children
[robot_state_publisher-4] Link rear_left_wheel_link had 0 children
[robot_state_publisher-4] Link rear_right_wheel_link had 0 children
[robot_state_publisher-4] [INFO] [1636879622.409395700] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1636879622.409787774] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1636879622.409879570] [robot_state_publisher]: got segment camera_depth_link
[robot_state_publisher-4] [INFO] [1636879622.409915903] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1636879622.409944107] [robot_state_publisher]: got segment front_left_wheel_link
[robot_state_publisher-4] [INFO] [1636879622.409971903] [robot_state_publisher]: got segment front_right_wheel_link
[robot_state_publisher-4] [INFO] [1636879622.409999181] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1636879622.410025106] [robot_state_publisher]: got segment laser
[robot_state_publisher-4] [INFO] [1636879622.410052477] [robot_state_publisher]: got segment rear_left_wheel_link
[robot_state_publisher-4] [INFO] [1636879622.410079606] [robot_state_publisher]: got segment rear_right_wheel_link
[ydlidar_ros2_driver_node-5] LiDAR successfully connected
[ekf_node-2] Warning: Invalid frame ID "base_footprint" passed to canTransform argument target_frame - frame does not exist
[... cut ...]
[joint_state_publisher-3] [INFO] [1636879624.333609838] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[ydlidar_ros2_driver_node-5] [YDLIDAR3]:Fixed Size: 720
[ydlidar_ros2_driver_node-5] [YDLIDAR3]:Sample Rate: 4K
[ydlidar_ros2_driver_node-5] [YDLIDAR]:Single Fixed Size: 450
[ydlidar_ros2_driver_node-5] [YDLIDAR]:Sample Rate: 4K
[ydlidar_ros2_driver_node-5] [YDLIDAR INFO] Current Sampling Rate : 4K
[ydlidar_ros2_driver_node-5] [YDLIDAR INFO] Now YDLIDAR is scanning ......

`

raidhu, 2021-11-14:
echo /odom/unfiltered
`header:
  stamp:
    sec: 1636880258
    nanosec: 801000000
  frame_id: odom
child_frame_id: base_footprint
pose:
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance:
  - 0.0010000000474974513
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0010000000474974513
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0010000000474974513
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance:
  - 9.999999747378752e-05
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 9.999999747378752e-05
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 9.999999747378752e-05
---
`
 /odom/filtered - I didn't get this topic

grassjelly, 2021-11-14:
do you have for the imu?

raidhu, 2021-11-14:
yes i do have imu

raidhu, 2021-11-14:
echo for imu

`header:
  stamp:
    sec: 1636882466
    nanosec: 221000000
  frame_id: imu_link
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: 0.011594448662086115
  y: 0.0
  z: 0.0
angular_velocity_covariance:
- 9.999999747378752e-06
- 0.0
- 0.0
- 0.0
- 9.999999747378752e-06
- 0.0
- 0.0
- 0.0
- 9.999999747378752e-06
linear_acceleration:
  x: 0.22273682593367994
  y: -0.7735913416836411
  z: 9.93693646020256
linear_acceleration_covariance:
- 9.999999747378752e-06
- 0.0
- 0.0
- 0.0
- 9.999999747378752e-06
- 0.0
- 0.0
- 0.0
- 9.999999747378752e-06
---
`

grassjelly, 2021-11-14:
Is the warning continuous? Or does it stop? If it stops, that warning shows up right before the imu/odometry is published. If you launch the slam launch file, do you have the same error?

raidhu, 2021-11-14:
no the warning from the efk stops.
but in slam launch file continue - the error shown in ros2 launch linorobot2_navigation slam.launch.py
 is continuously shown.

`[async_slam_toolbox_node-1] [INFO] [1636882967.764174681] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1636882967.536 for reason 'Unknown'
[async_slam_toolbox_node-1] [INFO] [1636882967.877698649] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1636882967.650 for reason 'Unknown'
[async_slam_toolbox_node-1] [INFO] [1636882968.275302731] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1636882967.764 for reason 'Unknown'
[async_slam_toolbox_node-1] [INFO] [1636882968.275689398] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1636882967.878 for reason 'Unknown'
`

grassjelly, 2021-11-14:
does it stop as well? Are you able to create a map?

raidhu, 2021-11-14:
nope

raidhu, 2021-11-14:
[image]

grassjelly, 2021-11-14:
as in a complete map?

raidhu, 2021-11-14:
no i cant able to move the base_footprint robot model in the rviz doesn't move.
u can see only odom node is moving.
[image]

grassjelly, 2021-11-14:
could you try adding these https://github.com/cra-ros-pkg/robot_localization/blob/ros2/params/ekf.yaml#L23-L27 to this https://github.com/linorobot/linorobot2/blob/master/linorobot2_base/config/ekf.yaml . 0.1 would be a good start

raidhu, 2021-11-14:
thankyou so much grassjelly now the ekf error gone.

the robot model in the rviz doesn't move.
the map looks uneven i think its something to do with laser pose urdf file.

[image]

grassjelly, 2021-11-14:
Great! Could you please share your updated ekf.yaml here? Would be useful for other users. Please file another issue if you’re having problems with mapping. Thanks

raidhu, 2021-11-14:
**updated ekf.yaml**

`ekf_filter_node:
    ros__parameters:
        frequency: 50.0
        two_d_mode: true
        publish_tf: true
 
        map_frame: map             
        odom_frame: odom            
        base_link_frame: base_footprint 
        world_frame: odom 
		transform_time_offset: 0.1
        transform_timeout: 0.1

        #x     , y     , z,
        #roll  , pitch , yaw,
        #vx    , vy    , vz,
        #vroll , vpitch, vyaw,
        #ax    , ay    , az
        odom0: odom/unfiltered
        odom0_config: [false, false, false,
                       false, false, false,
                       true, true, false,
                       false, false, true,
                       false, false, false]

        imu0: imu/data
        imu0_config: [false, false, false,
                      false, false, false,
                      false, false, false,
                      false, false, true,
                      false, false, false]`

closed 2021-11-14 by raidhu