linorobot/linorobot2 #10: robot model move faster then real world robot

raidhu, 2021-11-16:
which would be wrong is it motor calibration.

grassjelly, 2021-11-17:
Yes. Make sure you get these parameters right: https://github.com/linorobot/linorobot2_hardware/blob/master/config/lino_base_config.h#L50-L62

raidhu, 2021-11-17:
I give those parameters as per my bot parameter.
I get count per rev value by running sample.
Is encoder counts is responsible for robot movement in rviz.

raidhu, 2021-11-17:
`//define your robot' specs here
#define MOTOR_MAX_RPM 100                   // motor's max RPM          
#define MAX_RPM_RATIO 0.30                  // max RPM allowed for each MAX_RPM_ALLOWED = MOTOR_MAX_RPM * MAX_RPM_RATIO          
#define MOTOR_OPERATING_VOLTAGE 12          // motor's operating voltage (used to calculate max RPM)
#define MOTOR_POWER_MAX_VOLTAGE 13.2          // max voltage of the motor's power source (used to calculate max RPM)
#define MOTOR_POWER_MEASURED_VOLTAGE 12.9     // current voltage reading of the power connected to the motor (used for calibration)
#define COUNTS_PER_REV1 3899              // wheel1 encoder's no of ticks per rev
#define COUNTS_PER_REV2 3708              // wheel2 encoder's no of ticks per rev
#define COUNTS_PER_REV3 3844              // wheel3 encoder's no of ticks per rev
#define COUNTS_PER_REV4 3701              // wheel4 encoder's no of ticks per rev
#define WHEEL_DIAMETER 0.100                // wheel's diameter in meters
#define LR_WHEELS_DISTANCE 0.465            // distance between left and right wheels
#define PWM_BITS 10                          // PWM Resolution of the microcontroller
#define PWM_FREQUENCY 20000                 // PWM Frequency
`

raidhu, 2021-11-17:
fixed

closed 2021-11-17 by raidhu