linorobot/linorobot2 #15: Recovery Backup Not working

philmurp, 2021-12-22:
Hello, I am running Linorobot2 on Mecanum, NVIDIA Jetson computer with Teensy 4.1.

When my robot is stuck in a corner, I see it enters the recovery tree and does the spin and wait actions; however it has never performed the backup function and as a result it usually ends up aborting after a few tries and I have to restart the entire navigation stack.

1) Based on the Nav2 documentation it looks all pretty default for the parameters for recovery server so not sure why it wont try backup when it tries spin and wait pretty well

2) If it does abort is there any reason that I need to completely restart the entire Nav stack, rather than it being able to accept a new goal?

grassjelly, 2021-12-24:
Hi Phil,

This sounds more of a Nav2 related problem. Perhaps you could post in ROS answers instead.

philmurp, 2021-12-24:
Sure I will post over there.

grassjelly, 2022-03-05:
Closing this for now

closed 2022-03-05 by grassjelly