linorobot/linorobot2 #16: Is there anyway that i can adapt this for a 6 wheels "lookalike" mars rover?

Rossobimbo, 2022-01-15:
As per title i've almost finished the build of a 6 wheels rover, rocker bogie joint and the whole shabam.
Quite an interesting project, and after the failed SpotMini [which i'm planning to go back on sometime in the future] i think i'm very close to the finish line with this one.
The one thing though, as it is an adaptation i've decided to simplify it a bit, by not adding the 4 rotating front and back wheels, i've preferred a simpler setup.

I was very eager to make this project autonomous, and after finding linorobot, i thought it would have been easier, the problem is that you've only designed this for 2 or 4 wheels; is there a way to make it 6?

Appreciate any help that you can supply.

BR
Vittorio

grassjelly, 2022-01-16:
Hey Vittoria,

Yes this should be possible. You can perhaps start tweaking the kinmeatics library to have 6 wheels here https://github.com/linorobot/linorobot2_hardware/blob/master/firmware/lib/kinematics/kinematics.cpp

and create 6 instances for each of these objects https://github.com/linorobot/linorobot2_hardware/blob/master/firmware/src/firmware.ino#L60-L73, as well the related function calls.

Rossobimbo, 2022-01-23:
Hi grassjelly,

thanks a lot for your comment.

i've finally managed to get sometime to get around the 2 files to modify. 
I'm neither a programmer or a roboticist, but i think i managed to change all the values as per your suggestion.

###kinematics.cpp###
I do however have some issue in the file "kinematics.cpp" on line 122 and 129:

Line 122:
average_rps_y = ((float)(-rpm1 + rpm2 + rpm3 - rpm4) / total_wheels_) / 60.0; // RPM

Line 129:
average_rps_a = ((float)(-rpm1 + rpm2 - rpm3 + rpm4) / total_wheels_) / 60.0;

I'm not sure what sign + or - to insert with "rpm5" and "rpm6" particularly since now 5 and 6 are the 2 backwheels. [or perhaps rpm3 and rpm4 signs needs also change since these are now the middle wheels?!?]
Also in the switch statement, should i now add a "ROVER" case since all of the wheels on this case have their own motor, and there's 6 of them?

###firmware.ino###
i think i'm ok with this.

I would really appreciate if you could have a look at the modified .cpp file for me.

Thanks

Rossobimbo, 2022-01-24:
Sorry i forgot to upload the modified file. Github won't let me upload .cpp so i changed in .txt

Thanks
[kinematics.txt](https://github.com/linorobot/linorobot2/files/7924540/kinematics.txt)

grassjelly, 2022-01-29:
just make all wheels on the left (odd ids) negative and wheels on the right (even ids) positive. Average rps_y can be ignored

grassjelly, 2022-03-05:
Closing this for now. Feel free to open again if you have further questions

closed 2022-03-05 by grassjelly