linorobot/linorobot2 #18: Colcon Build issue

longwm, 2022-02-14:
Some assistance please.I am unable to compile during colcon build, I keep getting the following issue. Trying to install with zed2i ubuntu 20 on jetson nano image. Thank You

/home/jetson/dev_ws/src/image_common/camera_info_manager/src/camera_info_manager.cpp:46:10: fatal error: rcpputils/env.hpp: No such file or directory
 #include "rcpputils/env.hpp"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_info_manager.dir/build.make:63: CMakeFiles/camera_info_manager.dir/src/camera_info_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/camera_info_manager.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< camera_info_manager [5.59s, exited with code 2]
Aborted  <<< turtlesim [13.4s]                                                 
Aborted  <<< zed_components [7.67s]                            

Summary: 2 packages finished [14.5s]
  1 package failed: camera_info_manager
  2 packages aborted: turtlesim zed_components
  2 packages had stderr output: camera_info_manager zed_components
  3 packages not processed

grassjelly, 2022-02-15:
hey are you installing through the script?

longwm, 2022-02-15:
> hey are you installing through the script?

Hey, I tried both a manual install and using the script. No luck. Any suggestions?

longwm, 2022-02-21:
> hey are you installing through the script?

Hi, Any update on this please? I am still struggling to get it installed. same error every time, I tried changing out as per info on google group, assistance will be greatly appreciated. Thank you

grassjelly, 2022-03-05:
Thanks for posting this issue. Fixed this with https://github.com/linorobot/linorobot2/commit/2f6ee75f071028db8e49976bff4283a9b89906ee . $ROS_DISTRO branch on image_common has to be used to fix the issue.

grassjelly, 2022-03-12:
Closing this for now

closed 2022-03-12 by grassjelly