linorobot/linorobot2 #20: Mecanum and 4wd robots look just the same

Lindsay284, 2022-03-06:
I want to ask if the two urdf files are the same.... I felt no difference except the depth sensor pose....QAQ

grassjelly, 2022-03-06:
Appearance is the same, but the macanum's URDF uses this controller https://github.com/linorobot/linorobot2/blob/master/linorobot2_description/urdf/controllers/omni_drive.urdf.xacro so the robot can move holonomically.

Lindsay284, 2022-03-06:
OK, thanks for your anwsering. This helps me a lot.

> Appearance is the same, but the macanum's URDF uses this controller https://github.com/linorobot/linorobot2/blob/master/linorobot2_description/urdf/controllers/omni_drive.urdf.xacro so the robot can move holonomically.

closed 2022-03-12 by grassjelly