linorobot/linorobot2 #21: I encountered [ERROR] [waypoint_follower-8]: process has died [pid 57803, exit code 255, cmd '/opt/ros/galactic/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower --params-file /tmp/tmps86t6qm3 -r /tf:=tf -r /tf_static:=tf_static'].

Lindsay284, 2022-03-07:
When I use "ros2 launch linorobot2_navigation navigation.launch.py", it turns out this output     "[waypoint_follower-8] [FATAL] [1646652552.515721013] [waypoint_follower]: Can not get 'plugin' param value for wait_at_waypoint
[ERROR] [waypoint_follower-8]: process has died [pid 57803, exit code 255, cmd '/opt/ros/galactic/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower --params-file /tmp/tmps86t6qm3 -r /tf:=tf -r /tf_static:=tf_static']."
It seems that the params of waypoint_follower are wrong, I got this answer from google, this maybe because the navigation version is a little bit old when you develop......
I'm using navigation2 and ros2 galactic.

Lindsay284, 2022-03-07:
I've solved this problem by copy the newest nav2_params.yaml to navigation.yaml file. But I encountered a new error: `[controller_server-4] [INFO] [1646654953.669869078] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist`

crimean7, 2022-03-21:
Same issue here.

grassjelly, 2022-03-28:
@crimean7  @Lindsay284 please use the latest galactic branch.

closed 2022-04-15 by grassjelly