linorobot/linorobot2 #22: micro-ros issue in Galactic

crimean7, 2022-03-21:
Jetson Nano
ROS2 Galactic

Teensy goes to the error loop (led flashes two times) or, sometimes, micro-ros agent restarts when the robot receives the goal.

Seems there isn't such error in ROS2 Foxy.

I'll really appreciate an assistance to fix it.

grassjelly, 2022-03-21:
could you try this minimal example? https://github.com/linorobot/micro_ros_platformio

crimean7, 2022-03-21:
grassjelly, thanks for the prompt response.

The example is working as intended.

grassjelly, 2022-03-21:
strange, the Teensy should be able to reconnect. It usually takes 5 to 20 seconds. Try waiting for Teensy to fully reconnect before sending the Twist commands

crimean7, 2022-03-21:
Sorry, probably, I didn't explained it right.

When linorobot firmware is burned the Teensy reconnects by itself. But 'falling' into error loop (or restarts the micro-ros connection) once goal is set.

The example you sent the link is working as intended, but need to reconnect the teensy each time the micro-ros started.

---------
root@m1b3-t:/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1647875931.952952] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1647875931.953906] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1647875940.978777] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1647875940.988966] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1647875941.990627] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1647875942.992111] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1647875943.993621] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1647875944.533476] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1647875946.421842] info     | Root.cpp           | create_client            | create                 | client_key: 0x50C7822F, session_id: 0x81
[1647875946.423212] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x50C7822F, address: 0
[1647875946.477200] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x50C7822F, participant_id: 0x000(1)
[1647875946.483103] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x50C7822F, topic_id: 0x000(2), participant_id: 0x000(1)
[1647875946.488406] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x50C7822F, publisher_id: 0x000(3), participant_id: 0x000(1)
[1647875946.496285] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x50C7822F, datawriter_id: 0x000(5), publisher_id: 0x000(3)
-----
'device not found' is when I disconnect teensy

grassjelly, 2022-03-21:
by goal, do you mean Navigation goal?

crimean7, 2022-03-21:
teleop is working fine

closed 2022-04-03 by grassjelly

adityasivaraj, 2022-08-26:
Hi @crimean7 and @grassjelly , I am facing the same problem. Were you able to find the cause and solution?