linorobot/linorobot2 #24: Package 'micro_ros_agent' not found

Russ76, 2022-04-01:
The testing and motor running worked fine, but now I'm trying to get it to drive around, and this error comes up.

Galactic, Ubuntu 20.02, Raspi, Teensy 4.0, L298N motor controllers.

I still have it connected to screen, keyboard and mouse. Does that cause it?

Russ76, 2022-04-01:
I did get it to work, reinstalling several things from the manual installation script.
It worked for a bit, but all wheels weren't quite working together, with the joystick.
Then it quit. Ros software seems to still be working, LED flashes as if Teensy is getting commands, but no motors turn now. Will have to get out the multi-meter...

Russ76, 2022-04-02:
Now I'm having trouble with one machine seeing the topics from the other. Trying to run the joystick on laptop, and it isn't seen on robot, and robot's topics aren't seen on laptop. Machines can ping each other and ssh into each other, but topics don't share. Don't see anything about blocking UDP multicast in my router. ???

Russ76, 2022-04-03:
OK, we got robot to drive around! But it doesn't drive well. It hardly turns, goes forward and reverse better, but not slowly or with good control. Sometimes one wheel seems to want to run by itself! These l298n units are inexpensive and maybe aren't the best quality. I will try putting in another for the front one that seems to work the worst.

I couldn't get the robot (raspi) nodes to connect to laptop until I deleted the init-cloud-50 file and did wireless network using the desktop tools on Ubuntu. I had first made that file so that it could connect to wireless without the desktop installed, thinking I would just use the server version of Ubuntu 20.04. However, for many install routines and copying files, having the desktop available is a huge convenience.

It seems to want much more power for angular movements. How can we turn that up? On my own robot I had the angular boost set at 1.5 times the linear rate. For a skid-steer robot, I find that it drives much better that way.

Thanks for all your hard work, Juan!

Russ76, 2022-04-03:
Would you like photos of this little machine?

grassjelly, 2022-04-03:
hey @Russ76 would love to see the robot :) . Do you mind if I close this issue and you open a new one for the angular movement problem? Thank you!

closed 2022-04-03 by grassjelly