linorobot/linorobot2 #25: linorobot2_viz missing Galactic branch, and realsense issue

Sylvania2, 2022-04-04:
Hi

Installed Galactic last night :-)
git clone -b $ROS_DISTRO https://github.com/linorobot/linorobot2_viz src/linorobot2_viz fails.
removed -b $ROS_DISTRO  and it installs.
After Linorobot2 installation i found that realsense udev rules was missing, downloaded and fixed.
Now Robot runs perfectly as a 2wd with rplidar and also with realsense as lidar.
But using both rplidar and realsense only lidar seems to be used.
Rviz shows no data in "bumper hit" , is there any tricks to get voxel layer / pointcloud working ?

Best regards

Sylvania2, 2022-04-08:
Hi grassjelly

file linorobot2_navigation/config/navigation.yaml has this setup:

realsense:
          topic: /camera/depth/color/points

That topic is missing when running bringup and navigation.

ros2 launch linorobot2_bringup bringup.launch.py
ros2 launch linorobot2_navigation navigation.launch.py
ros2 topic list
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu

Best regards

Sylvania2, 2022-04-15:
Hi

Tested Galactic/Linorobot2 on RPI4, Jetson Nano and Celeron. No data on topic /camera/depth/color/points, seems to be a realsense issue when using arm, celeron was working.
Foxy is working with data on /camera/depth/color/points . Resolution and 6fps on Jetson and Rpi4.

Best regards

grassjelly, 2022-04-15:
Thanks @Sylvania2 . Just realized this recently too. The same settings worked on Foxy but missing /points topic on Galactic. Would you wanna file an Issue on Realsense repo itself? I'll close this for now

closed 2022-04-15 by grassjelly

Sylvania2, 2022-04-20:
Hi

Changed depth.lauch.py to lower resolution and newer parameter setup.
Also changed to correct topics in rviz !  working :-)

`        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(PathJoinSubstitution(
                [FindPackageShare('realsense2_camera'), 'launch', 'rs_launch.py']
            )),
            condition=LaunchConfigurationEquals('sensor', 'realsense'),
            launch_arguments={
                'pointcloud.enable': 'true',
                'depth_module.profile': '424x240x6',
                'rgb_camera.profile': '424x240x6',
                'ordered_pc': 'true',
                'pointcloud_texture_stream' : 'RS2_STREAM_ANY',
                'initial_reset': 'true'
            }.items()
`

Best regards
Kim