linorobot/linorobot2 #26: Mapping not stable in Linorbot

vinayaksandilya, 2022-04-07:
[image]
My robot is working normally but even after modify eky.yaml and all suggested problems i am not able receive the map properly 
```
Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rviz2-1] Warning: Invalid frame ID "camera_depth_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
```

in ros2 launch linorobot2_viz slam.launch.py
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-04-07-17-29-39-629182-ubuntu-5607
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [5609]
[async_slam_toolbox_node-1] [INFO] [1649352580.291677050] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1649352580.769956287] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1649352580.773386071] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] [WARN] [1649352580.819195176] [slam_toolbox]: maximum laser range setting (20.0 m) exceeds the capabilities of the used Lidar (12.0 m)
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
```
in ros2 launch linorobot2_navigation slam.launch.py
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-04-07-17-28-15-296529-ubuntu-5247
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [micro_ros_agent-1]: process started with pid [5250]
[INFO] [ekf_node-2]: process started with pid [5252]
[INFO] [joint_state_publisher-3]: process started with pid [5254]
[INFO] [robot_state_publisher-4]: process started with pid [5256]
[INFO] [rplidar_composition-5]: process started with pid [5258]
[micro_ros_agent-1] [1649352497.176002] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[micro_ros_agent-1] [1649352497.176696] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[rplidar_composition-5] [INFO] [1649352497.237481509] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
[robot_state_publisher-4] Link base_link had 7 children
[robot_state_publisher-4] Link camera_link had 1 children
[robot_state_publisher-4] Link camera_depth_link had 0 children
[robot_state_publisher-4] Link front_left_wheel_link had 0 children
[robot_state_publisher-4] Link front_right_wheel_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link laser had 0 children
[robot_state_publisher-4] Link rear_left_wheel_link had 0 children
[robot_state_publisher-4] Link rear_right_wheel_link had 0 children
[robot_state_publisher-4] [INFO] [1649352497.270687730] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1649352497.271041504] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1649352497.271124707] [robot_state_publisher]: got segment camera_depth_link
[robot_state_publisher-4] [INFO] [1649352497.271164040] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1649352497.271197021] [robot_state_publisher]: got segment front_left_wheel_link
[robot_state_publisher-4] [INFO] [1649352497.271229687] [robot_state_publisher]: got segment front_right_wheel_link
[robot_state_publisher-4] [INFO] [1649352497.271260891] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1649352497.271293742] [robot_state_publisher]: got segment laser
[robot_state_publisher-4] [INFO] [1649352497.271324779] [robot_state_publisher]: got segment rear_left_wheel_link
[robot_state_publisher-4] [INFO] [1649352497.271356297] [robot_state_publisher]: got segment rear_right_wheel_link
[micro_ros_agent-1] [1649352498.157387] info     | Root.cpp           | create_client            | create                 | client_key: 0x5E218E29, session_id: 0x81
[micro_ros_agent-1] [1649352498.157540] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5E218E29, address: 0
[micro_ros_agent-1] [1649352498.200126] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x5E218E29, participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.205602] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5E218E29, topic_id: 0x000(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.210683] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x5E218E29, publisher_id: 0x000(3), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.216667] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x5E218E29, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[micro_ros_agent-1] [1649352498.222560] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5E218E29, topic_id: 0x001(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.227583] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x5E218E29, publisher_id: 0x001(3), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.233474] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x5E218E29, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[micro_ros_agent-1] [1649352498.238540] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x5E218E29, topic_id: 0x002(2), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.243687] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x5E218E29, subscriber_id: 0x000(4), participant_id: 0x000(1)
[micro_ros_agent-1] [1649352498.249762] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x5E218E29, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[joint_state_publisher-3] [INFO] [1649352498.813956023] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[rplidar_composition-5] [INFO] [1649352499.761990349] [rplidar_composition]: RPLIDAR S/N: 7FF6EDF9C7E29BCEA7E39EF2EF254304
[rplidar_composition-5] [INFO] [1649352499.762214995] [rplidar_composition]: Firmware Ver: 1.29
[rplidar_composition-5] [INFO] [1649352499.762313624] [rplidar_composition]: Hardware Rev: 7
[rplidar_composition-5] [INFO] [1649352499.763890090] [rplidar_composition]: RPLidar health status : '0'
[rplidar_composition-5] [INFO] [1649352500.307478918] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2, flip_x_axis 0

```
in ros2 launch linorobot2_bringup bringup.launch.py

vinayaksandilya, 2022-04-07:
and in ros2 topic list 
`/clicked_point
/cmd_vel
/diagnostics
/goal_pose
/imu/data
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/odom
/odom/unfiltered
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static
`

grassjelly, 2022-04-07:
are you able to echo /odom/unfiltered?

vinayaksandilya, 2022-04-07:
yes 
I am receiving the data from it

vinayaksandilya, 2022-04-07:
here the output 
```
header:
  stamp:
    sec: 1649357456
    nanosec: 520000000
  frame_id: odom
child_frame_id: base_footprint
pose:
  pose:
    position:
      x: -0.685720682144165
      y: -0.1940639168024063
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.8188973665237427
      w: 0.5739399194717407
  covariance:
  - 0.0010000000474974513
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0010000000474974513
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0010000000474974513
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance:
  - 9.999999747378752e-05
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 9.999999747378752e-05
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 9.999999747378752e-05
---
```

vinayaksandilya, 2022-04-08:
Solved it silly issue i put the lidar the wrong way motor to the back 
[image]

closed 2022-04-08 by vinayaksandilya