linorobot/linorobot2 #27: YDlidar driver error: name and namespace

Russ76, 2022-04-12:
ubuntu@ubuntu:~$ ros2 launch ydlidar_ros2_driver ydlidar_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-04-11-19-57-04-174823-ubuntu-3782
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/ubuntu/lino3_ws/install/ydlidar_ros2_driver/share/ydlidar_ros2_driver/launch/ydlidar_launch.py", line 38, in generate_launch_description
    driver_node = LifecycleNode(package='ydlidar_ros2_driver',
TypeError: __init__() missing 2 required keyword-only arguments: 'name' and 'namespace'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 127, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: __init__() missing 2 required keyword-only arguments: 'name' and 'namespace'
ubuntu@ubuntu:~$ 

This attempt was with screen and keyboard connected. Raspberry Pi and Ubuntu 20.04

Russ76, 2022-04-12:
Correction: The above was running the YDlidar by itself.
With the whole Linorobot2 package, it started spinning!
and:
ubuntu@ubuntu:~$ ros2 topic list -v
Published topics:
 * /diagnostics [diagnostic_msgs/msg/DiagnosticArray] 1 publisher
 * /imu/data [sensor_msgs/msg/Imu] 1 publisher
 * /joint_states [sensor_msgs/msg/JointState] 1 publisher
 * /odom [nav_msgs/msg/Odometry] 1 publisher
 * /odom/unfiltered [nav_msgs/msg/Odometry] 1 publisher
 * /parameter_events [rcl_interfaces/msg/ParameterEvent] 5 publishers
 * /robot_description [std_msgs/msg/String] 1 publisher
 * /rosout [rcl_interfaces/msg/Log] 6 publishers
 * /scan [sensor_msgs/msg/LaserScan] 1 publisher
 * /tf [tf2_msgs/msg/TFMessage] 2 publishers
 * /tf_static [tf2_msgs/msg/TFMessage] 1 publisher

Subscribed topics:
 * /cmd_vel [geometry_msgs/msg/Twist] 1 subscriber
 * /imu/data [sensor_msgs/msg/Imu] 1 subscriber
 * /joint_states [sensor_msgs/msg/JointState] 1 subscriber
 * /odom/unfiltered [nav_msgs/msg/Odometry] 1 subscriber
 * /parameter_events [rcl_interfaces/msg/ParameterEvent] 4 subscribers
 * /robot_description [std_msgs/msg/String] 1 subscriber
 * /set_pose [geometry_msgs/msg/PoseWithCovarianceStamped] 1 subscriber
 * /tf [tf2_msgs/msg/TFMessage] 1 subscriber
 * /tf_static [tf2_msgs/msg/TFMessage] 1 subscriber

Russ76, 2022-04-12:
I had to edit the Ydlidar params file, because it wanted to use ttyACM0 as well as the Teensy, and this caused a conflict. After Teensy is connected, Ydlidar uses /dev/ydlidar
So I just inserted that info into params file, and poof- she works.

grassjelly, 2022-04-13:
thanks for reporting the issue. Fixed here: https://github.com/linorobot/linorobot2/commit/f60961c1eba72c6fde1049f7cb41274cf4e7d04d closing this

closed 2022-04-13 by grassjelly

Lucyliuwen, 2024-03-18:
I still encountered this problem, but I didn't find a solution, so how can you solve it?

Lucyliuwen, 2024-03-18:
> thanks for reporting the issue. Fixed here: [f60961c](https://github.com/linorobot/linorobot2/commit/f60961c1eba72c6fde1049f7cb41274cf4e7d04d) closing this

hello, I recheck the file attached by you, but it did not solve it.