linorobot/linorobot2 #28: Using ODrive

qiwalker, 2022-04-27:
Hi,

is it possible to use ODrive driver with linorobot ?

Thankyou..

Sylvania2, 2022-04-27:
Hi

ODRIVE Interface:
Servo PWM -- RC Recievers, Arduino, etc.

I am using VESC4 and they are working fine.

Best regards

qiwalker, 2022-04-27:
Hi, 

thanks for your reply, so is it possible to use with Linorobot with teensy ?
btw do you have partnumber for your VESC4 ?

here's my motor, will it work with VESC4 ?, i already try using Odrive and it working.
[image]

thankyou

Sylvania2, 2022-04-27:
Just google vesc4 or a clone like fsesc4..
https://flipsky.net/products/mini-fsesc4-20-50a-base-on-vesc-widely-used-in-eskateboard-escooter-ebike

Best regards

qiwalker, 2022-05-03:
hi, anyone now how to read hall sensor as an encoder ?
brushless motor already include the hall sensor for encoder, but i have no idea how to read this as a encoder output

thanks

grassjelly, 2022-05-19:
closing this for now. Please open a separate issue on linorobot2_hardware on hardware related questions. Thanks!