linorobot/linorobot2 #30: Realsense Pointcloud2 and Voxellayer issue

Sylvania2, 2022-06-07:
Voxellayer never adds marked voxels, and robot hits everthing lower than the laser. Pointcloud is working and can be seen in RVIZ
I tried changing raytrace_min_range: 0.5  to lower and higher value. But that didn't work

Found a suggestion that the sensor frame is wrong. Pointcloud topic: /camera/depth/color/points has "camera_depth_optical_frame" as frame_id. And that should be defined. Not tested yet.

`        realsense:
          topic: /camera/depth/color/points
          raytrace_min_range: 0.5
          min_obstacle_height: 0.0
          max_obstacle_height: 2.0 
          clearing: True
          marking: True
          sensor_frame:  camera_depth_optical_frame
          data_type: "PointCloud2"`

Or that the depth_sensor.urdf.xacro should match the Pointcloud frame_id:
`    <link name="camera_depth_link"/>
    <joint name="camera_depth_joint" type="fixed">
      <origin xyz="0 0 0" rpy="-1.57 0 -1.57"/>
      <parent link="camera_link"/>
      <child link="camera_depth_link"/>
    </joint>
`
And use camera_depth_optical_frame"
`  <joint name="camera_depth_optical_joint" type="fixed">
    <origin rpy="-1.57079632679 0 -1.57079632679" xyz="0 0 0"/>
    <parent link="camera_depth_frame"/>
    <child link="camera_depth_optical_frame"/>
  </joint>
  <link name="camera_depth_optical_frame">
`

Anyone using RpLidar and Realsense Pointcloud2 / Voxellayer ? And how did you make this work ?

Best regards Kim

grassjelly, 2022-06-08:
hey @Sylvania2 . Thanks for reporting this issue. It was actually working just that it's only available on the global costmap. Added the same parameter on local costmap. You can check out the changes here: https://github.com/linorobot/linorobot2/commit/08401ddb5d590ae1699edec3788ecf708c14b4b4

Sylvania2, 2022-06-08:
Fix is working. Thanks. Closing

closed 2022-06-08 by Sylvania2