linorobot/linorobot2 #31: [ERROR] [joint_state_publisher-3]: process has died

Sylvania2, 2022-06-08:
Error while launching bringup:

`[joint_state_publisher-3] Traceback (most recent call last):
[joint_state_publisher-3]   File "/opt/ros/galactic/lib/joint_state_publisher/joint_state_publisher", line 11, in <module>
[joint_state_publisher-3]     load_entry_point('joint-state-publisher==2.2.0', 'console_scripts', 'joint_state_publisher')()
[joint_state_publisher-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher/joint_state_publisher.py", line 415, in main
[joint_state_publisher-3]     jsp = JointStatePublisher(parsed_args.urdf_file)
[joint_state_publisher-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/joint_state_publisher/joint_state_publisher.py", line 217, in __init__
[joint_state_publisher-3]     super().__init__('joint_state_publisher', automatically_declare_parameters_from_overrides=True)
[joint_state_publisher-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 216, in __init__
[joint_state_publisher-3]     self.declare_parameters(
[joint_state_publisher-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 487, in declare_parameters
[joint_state_publisher-3]     raise ParameterAlreadyDeclaredException(parameters_already_declared)
[joint_state_publisher-3] rclpy.exceptions.ParameterAlreadyDeclaredException: ('Parameter(s) already declared', ['use_sim_time'])
[ERROR] [joint_state_publisher-3]: process has died [pid 11733, exit code 1, cmd '/opt/ros/galactic/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher --params-file /tmp/launch_params_5pd3ox71'].
`

https://githubplus.com/ros2/rclpy/issues/948

Error is caused by the parameters "use_sim_time" in desciption.launch.py,:

`        Node(
            package='joint_state_publisher',
            executable='joint_state_publisher',
            name='joint_state_publisher',
            condition=IfCondition(LaunchConfiguration("publish_joints")),
            parameters=[
                {'use_sim_time': LaunchConfiguration('use_sim_time')}
            ]
        ),

`
Removing
            ,parameters=[
                {'use_sim_time': LaunchConfiguration('use_sim_time')}
            ]

Fixes the ERROR but might not be the correct way to fix this issue.

Best regards Kim

grassjelly, 2022-06-08:
thanks for reporting this issue. This has been fixed here https://github.com/linorobot/linorobot2/commit/0aad8a4ce1f0ad503b3cd1649a7ee4e217b72ce4 . Please use an updated version

grassjelly, 2022-06-08:
Sorry, just realised I closed this. Please close this issue if the commit resolves your issue @Sylvania2 . Thanks!

closed 2022-06-08 by Sylvania2