linorobot/linorobot2 #33: slam toolbox, RobotModel Status : Error

hwacoding, 2022-07-04:
$  ros2 launch linorobot2_description description.launch.py rviz:=true
I can see my robot.

But, When I doing slam.
$ ros2 launch linorobot2_navigation slam.launch.py rviz:=true sim:=true
[image]
[image]

There are no RobotModel base_footprint and base_link else.
But, I can see these when I 
$  ros2 launch linorobot2_description description.launch.py rviz:=true

Also, I  [slam_toolbox]: Message Filter dropping message: frame 'laser'
So I try follow this.

Try to up transform_timeout by 0.1 in linorobot2_navigation/config/slam.yaml until the warning is gone.

But, I this Message keep showing.

Please help me.

grassjelly, 2022-07-04:
hey @hwacoding. Could you reply this pls. https://github.com/linorobot/linorobot2_hardware/issues/13

hwacoding, 2022-07-04:
Ok. sorry. I reply it now.

hwacoding, 2022-07-04:
[image]
/robot_description topic have to go /ekf_filter_node?

grassjelly, 2022-07-04:
Don't use the sim argument if you're mapping using the real robot. 
`ros2 launch linorobot2_navigation slam.launch.py rviz:=true`

closed 2022-07-06 by hwacoding