linorobot/linorobot2 #35: Install Robot fails w/ humble

JHPHELAN, 2022-07-09:
Following
[https://github.com/linorobot/linorobot2#installation](url):
```
ubuntu@LINOROBOT:/tmp$ echo $ROS_DISTRO
humble
ubuntu@LINOROBOT:~$ echo $(rosversion -d)
humble
ubuntu@LINOROBOT:~$ source /opt/ros/humble/setup.bash
ubuntu@LINOROBOT:~$ cd /tmp
ubuntu@LINOROBOT:/tmp$ wget https://raw.githubusercontent.com/linorobot/linorobot2/${ROS_DISTRO}/install_linorobot2.bash
--2022-07-09 18:10:16--  https://raw.githubusercontent.com/linorobot/linorobot2/humble/install_linorobot2.bash
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8000::154, 2606:50c0:8001::154, 2606:50c0:8002::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8000::154|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-07-09 18:10:16 ERROR 404: Not Found.

ubuntu@LINOROBOT:/tmp$ wget https://raw.githubusercontent.com/linorobot/linorobot2/humble/install_linorobot2.bash
--2022-07-09 18:18:25--  https://raw.githubusercontent.com/linorobot/linorobot2/humble/install_linorobot2.bash
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 2606:50c0:8001::154, 2606:50c0:8003::154, 2606:50c0:8002::154, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|2606:50c0:8001::154|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-07-09 18:18:25 ERROR 404: Not Found.

```

grassjelly, 2022-07-10:
hey @JHPHELAN . Just created a humble branch. Could you try again?

JHPHELAN, 2022-07-10:
@grassjelly 
Thanks for the very prompt responses!
Downloaded ok.  Then
`ubuntu@LINOROBOT:~$ bash install_linorobot2.bash 4wd realsense realsense`

`Summary: 1 package finished [7.98s]
EE
=== ./uros/micro-ROS-Agent (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/micro-ROS/micro-ROS-Agent.git/': Failed to connect to github.com port 443 after 27 ms: Network is unreachable
=== ./uros/micro_ros_msgs (git) ===
Could not determine ref type of version: fatal: unable to access 'https://github.com/micro-ROS/micro_ros_msgs.git/': Failed to connect to github.com port 443 after 27 ms: Network is unreachable
[ros2run]: Process exited with failure 1`

Does this have anything to do with the caveat about skipping micro-ROS keys when running rosdep install?
Is micro-ROS necessary to this project?

grassjelly, 2022-07-18:
hey, sorry need to find bandwidth to take a look at this as ROS2 Humble is not currently supported yet.

JHPHELAN, 2022-07-21:
Any progress?  Should I try again?

JHPHELAN, 2022-07-22:
Running: RPi4B 8GB / Ubuntu 22.04 Jammy Jellyfish / ROS2 Humble Hawksbill
trying to be sure I did all the correct steps I did:
`ubuntu@LINOROBOT:~$ rm -rf ~/linorobot2_ws/src/micro_ros_setup/

ubuntu@LINOROBOT:~/linorobot2_ws$ rosdep update && rosdep install --from-path src --ignore-src -y --skip-keys microxrcedds_agent --skip-keys micro_ros_agent

ubuntu@LINOROBOT:~/linorobot2_ws$ colcon build

ubuntu@LINOROBOT:~$ source /opt/ros/humble/setup.bash

ubuntu@LINOROBOT:~$ bash install_linorobot2.bash 4wd realsense realsense
#All required rosdeps installed successfully
Building micro-ROS Agent
**Package 'micro_ros_agent' specified with --packages-up-to was not found**
[ros2run]: Process exited with failure 1
`
I looked at the install_linorobot2.bash file but didn't see  "--packages-up-to"  anywhere.
Otherwise it seems to be making progress

grassjelly, 2022-07-24:
hey I'm sorry for the delay. I'm waiting for a stable docker image (w/ VNC) for Humble then i can test the whole package. I'll post it here once the migration is ready.

samuk, 2023-01-16:
Is this what you need? https://github.com/osrf/docker_images/tree/master/ros/humble/ubuntu/jammy

closed 2023-08-30 by grassjelly