linorobot/linorobot2 #38: Gazebo install fails on RPi / U22 Jammy / ROS2 Humble

JHPHELAN, 2022-07-31:
ubuntu@LINOROBOT:/tmp$ **bash install_linorobot2.bash 4wd ydlidar realsense**
...
E: Unable to locate package ros-humble-gazebo-ros-pkgs
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros-pkgs] failed

Trying directly:
ubuntu@LINOROBOT:/tmp$ **sudo -H apt-get install -y ros-humble-gazebo-ros-pkgs**
also fails

according to [https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1158](url)
chapulina commented on Sep 9, 2020
Board: Raspberry Pi4 4Gb of RAM
ros foxy on a Rpi4
**This is the problem, gazebo-ros-pkgs is only available for amd64:**

Confirmed by looking in:
[http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-gazebo-ros-pkgs/](url)

ros-<distro>-desktop-full is reported to contain gazebo.  It is NOT available.  
whereas ros-<distro>-desktop does not.  IS available for humble.
**Need a work-around to install gazebo under ROS2 Humble / U22 Jammy  / Raspberry Pi 4B-8GB**

samuk, 2022-08-04:
Yeah I just ran into this too 

ubuntu@ubuntu:~$ sudo -H apt-get install -y ros-humble-gazebo-ros-pkgs
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-gazebo-ros-pkgs

samuk, 2022-08-05:
This looks relevant: https://github.com/osrf/gazebo/issues/3236#issuecomment-1204137725

samuk, 2022-08-08:
This has been fixed upstream and I can now install using the PPA listed here: https://github.com/osrf/gazebo/issues/3236

samuk, 2022-12-29:
It looks like ros-humble-gazebo-ros-pkgs will never exist for Arm. Workarounds here: https://github.com/gazebosim/gazebo-classic/issues/3236#issuecomment-1331053760

grassjelly, 2023-01-01:
Hey @samuk I don't think it's also ideal to run Gazebo on PI anyway.

closed 2023-08-30 by grassjelly