linorobot/linorobot2 #45: "Odom frame not found" when launching navigation and joint state publisher stuck on waiting to publish robot_description topic

zyongye, 2022-08-31:
Hi, I'm following the quick start linorobot2 and run exact the same install command except replace "micro-ros-agent" to my own node to process "/cmd_vel" topics. However, the following two message keeping showing up and the rviz keeping showing no transform from all part to "map" topics. 

When I run nav2 package, 

[image]

When I run bring_up. 

[image]

[image]

Any help is appreciated!!!

Thanks in advance!!!

grassjelly, 2022-09-01:
can you confirm the following? https://github.com/linorobot/linorobot2_hardware#3-check-the-topics

zyongye, 2022-09-01:
Yeah they are all present.

zyongye, 2022-09-01:
Also I only use laser sensor...

grassjelly, 2022-09-01:
are you able to echo odom topic?

zyongye, 2022-09-01:
Yeah there's no any other output..

grassjelly, 2022-09-04:
could you post the issue on linorobot2_hardware instead?

closed 2022-09-05 by zyongye

thanhtruongphan, 2024-11-01:
excuse me, @zyongye did you solve it? pls