linorobot/linorobot2 #48: No transform from [...] to [map]

EnderDragonEP, 2022-09-21:
I run:
`ros2 launch linorobot2_bringup bringup.launch.py`
`ros2 run teleop_twist_keyboard teleop_twist_keyboard`
`ros2 launch linorobot2_navigation slam.launch.py rviz:=true`
to slam, but I the transform wasn't working. And rviz has a lot of errors displayed all related to transform.
[image]
[image]
[image]

I'm new to ROS2 and wasn't able to Identify and fix the problems... 

URDF looks ok and working:
[image]

Topic list:
```
/clicked_point
/cmd_vel
/diagnostics
/goal_pose
/imu/data
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/odom
/odom/unfiltered
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static
```

System info:
System config: 2wd
System: Jetson nano 
OS: Ubuntu 20.04 from [Qengineering](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image)
Base board: Teency 4.0
Lidar: Rplidar
IMU:  MPU-9250
Motor driver: L298N

grassjelly, 2022-09-21:
Have you initialized the robot’s pose in rviz?

EnderDragonEP, 2022-09-21:
> Have you initialized the robot’s pose in rviz?
Thanks for the advice! I'll try.
But  SLAM need the position of the robot? That's kinda different from Linorobot revision one isn't it?

grassjelly, 2022-09-22:
Sorry, I missed the slam part. Are you able to confirm that these topics https://github.com/linorobot/linorobot2_hardware#3-check-the-topics are available and publishing topics?

EnderDragonEP, 2022-09-22:
> Sorry, I missed the slam part. Are you able to confirm that these topics https://github.com/linorobot/linorobot2_hardware#3-check-the-topics are available and publishing topics?

Yes they are available odometry and IMU data are present.
[image]

/odom/unfiltered
[image]

/imu/data
[image]

grassjelly, 2022-09-22:
Silly question, how about the `/scan` topic?

EnderDragonEP, 2022-09-22:
# MarkDown

> Silly question, how about the `/scan` topic?

Thank you for the head-up! I have found the cause of the problem. 
I was thinking about where is the laser scan lines goes and I was checking the [manual installation](https://github.com/linorobot/linorobot2/blob/galactic/ROBOT_INSTALLATION.md).
I installed the **Rplidar** and **Realsense** drivers and add the environment variables to the `.bashrc`.
Out of curiosity, I opened the `.bashrc` and I find that the  `LINOROBOT2_DEPTH_SENSOR` and `LINOROBOT2_LASER_SENSOR` variables are not set.
[image]

After I eddit the `.bashrc` and run the command `source ~/.bashrc`. The issue was fixed! No more errors.
[image]

But seems like the Intel Realsense camera still have some issue. But that the problem for another day. 😭
[image]

Thank for you help and guidance!

closed 2022-09-22 by EnderDragonEP