linorobot/linorobot2 #64: No transform from [ALL_WHEELS] to map

JoaquinUrrisa, 2023-03-23:
## Issue Summary

Hi, I'm new to ROS2 and I'm trying to run the mecanum robot simulated in gazebo and with SLAM.  However I get an error when trying to see the RobotModel in RViz.

## Issue Details

I followed the README, first using the `linorobot2_desciption` launch and i could see the RobotModel perfectly in Rviz. Then i used the `linorobot_gazebo launch` and everithing worked (i even moved the robot with the teleop package). But when i tried `linorobot_navigation slam.launch.py` with `rviz:=true` i got the following warning:
```
[rviz2-2] Warning: Invalid frame ID "front_left_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "front_right_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "rear_left_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "rear_right_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp

```
and the following error in Rviz:
[image]

## Steps to Reproduce
1. In one terminal I run `ros2 launch linorobot2_gazebo gazebo.launch.py`
2. Then, in another terminal I run `ros2 launch linorobot2_navigation slam.launch.py rviz:=true sim:=true`

## System Information

- Operating System: e.g. Ubuntu 22.04
- ROS Version: ROS2 Humble
- Linorobot2 Version: linorobot2 (humble branch)

## Additional Information

Despite the error the mapping works.

I noticed that the TF tree did not have tfs for the wheels. Upon some reasearch i found that usually the joint_state_publisher node is used for that purpose. So I changed  the `publish_joints` argument from `false` to `true` in:
```
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(description_launch_path),
            launch_arguments={
                'use_sim_time': str(use_sim_time),
                'publish_joints': 'false',
            }.items()
        ),
```
 in  the `gazebo.launch.py` file and was able to see the robot model in Rviz  (i also had to change the fixed frame from map to base_link despite the TF tree being as shown below).
 
[image]

grassjelly, 2023-03-23:
there's no TF between map frame and odom frame at runtime, you have to set the robot's initialpose (either from Rviz or programmatically https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L109) to give the robot its prior

JoaquinUrrisa, 2023-03-23:
Hi, thanks for the quick response! 

I tried setting the initial pose (both with Rviz 2D pose estimate tool and with 'ros topic pub /set_pose') but got the same result.

Can you explain a bit more why there's no TF between map and odom? Because without me setting the initialpose, the TF Tree already shows a TF from map to odom:

[image]

grassjelly, 2023-03-31:
sorry my bad, setting initialpose is only for navigation. Did you manage to create a full map even with those errors?

JoaquinUrrisa, 2023-04-10:
Yes, mapping works fine just two errors: 

- can't see the robot model in Rviz (due to ```Invalid frame ID``` error mentioned in this issue)

- ```[async_slam_toolbox_node-1] [INFO] [1681137581.820028677] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 198.001 for reason 'discarding message because the queue is full'```