linorobot/linorobot2 #65: Humble install fails - rosdep microros

brianlmerritt, 2023-04-04:
A couple points on the install

1. user should run any sudo command before the bash install_linorobot2.bash xwd etc etc to make sure the sudo password is cached
2. user should also run sudo rosdep update if they haven't already loaded that, or the install fails part way through
3. the rosdep update script fails with 

`updated cache in /home/blm/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
linorobot2_bringup: Cannot locate rosdep definition for [micro_ros_agent]`

This was on a fairly vanilla ros2 humble install Ubuntu 22.04 AMD64.  Due to the previous fails on steps 1 & 2, it may be that running the install again was an issue but I had deleted all previous workspaces and the repository directory.

stalinADAQ, 2023-04-13:
updated cache in /home/blm/.ros/rosdep/sources.cache ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies: linorobot2_bringup: Cannot locate rosdep definition for [micro_ros_agent]

I am facing same problem, how to resolve that ?

grassjelly, 2023-04-13:
there’s no rosdep keys for micro ros agent you use the skip flag to omit those warnings https://github.com/linorobot/linorobot2#21-install-linorobot2-package

stalinADAQ, 2023-04-13:
Thanks for quick reply, But i am trying to follow the ROBOT installation document. where we need install micro_ros_setup right ? if we skip, it is skipping micro_ros_setup installation itself ? can you advice ?

stalinADAQ, 2023-04-14:
@grassjelly Finally, I am able to install the ROBOT computer with install script easily. But now I have strange problem.
1. I have run the brinup script from the robot, and SLAM from host computer. The ROBOT is visible in the RVIZ.
2. but, it is very noisy or flickering the pose (I mean ROBOT is flickering on z-axis) even after ROBOT is stopped.
3. Becuase of this the entire SLAM is looking like a circle, please help, if the IMU is faulty or LIDAR has problem ?
[image]

stalinADAQ, 2023-04-15:
@grassjelly One more point, actually the Teensy firmware was compiled in Foxy setup, but recently i have migrated the ROBOT computer with Humble Setup. But still the old Foxy Teensy firmware will work together with Humble? or is it necessary to recompile the Teensy under Humble as well ?

stalinADAQ, 2023-04-15:
Please check attached video also