linorobot/linorobot2 #69: Humble, ros2 launch nav2_bringup navigation_launch.py errors

Sylvania2, 2023-05-02:
Missing map and other errors (Invalid frame ID “map” passed to canTransform argument target_frame - frame does not exist at line 93 in ./src/buffer_core.cpp) when running ros2 launch nav2_bringup navigation_launch.py

FIX: 
change robot_model_type: "differential" in navigation.yaml to robot_model_type: "nav2_amcl::DifferentialMotionModel"

Best regards
Kim

Sylvania2, 2023-09-25:
Hi

Last fix has been working until i did a "sudo apt update && sudo apt upgrade" and disabled the firewall ..

ros2 launch linorobot2_navigation navigation.launch.py rviz:=true output:

> [component_container_isolated-1] [INFO] [1695657995.999423102] [amcl]: Message Filter dropping message: frame 'laser' at time 1695657994.871 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1695657996.127430455] [amcl]: Message Filter dropping message: frame 'laser' at time 1695657994.999 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-2] [INFO] [1695657996.143159982] [rviz2]: Message Filter dropping message: frame 'laser' at time 1695657993.863 for reason 'discarding message because the queue is full'
[rviz2-2] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 93 in ./src/buffer_core.cpp
> 

errors in rviz:

[image]

grassjelly, 2023-09-25:
please use the latest humble branch. the amcl config has been updated