linorobot/linorobot2 #70: linorobot2 install bash error ( Galactic )

Sylvania2, 2023-05-07:
Going back to Galactic i get this error when running install script.
bash install_linorobot2.bash 2wd rplidar realsense ( Ubuntu 20.0.6 ROS2 Galactic ) 

(intel@intel-desktop:~/linorobot2_ws$ rosdep install --from-path src --ignore-src -y)
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
linorobot2_description: Cannot locate rosdep definition for [xacro]
linorobot2_navigation: Cannot locate rosdep definition for [nav2_bringup]
linorobot2_bringup: Cannot locate rosdep definition for [joy_linux]

so far i solved it with adding: rosdep update --include-eol-distros

Best regards

SHABRAWii, 2024-03-23:
I am facing the same issue.
Any Update?

SHABRAWii, 2024-03-23:
## **Problem**
I found out that these packages are **not** official **ros package** so you **can't install** them **with** **rosdep**. 
## **Solution**
Instead, you should install it with `apt-get` using `sudo apt-get install ros-<your-distro>-<package>`
Example: `sudo apt-get install ros-foxy-xacro`

**You will also face some errors about missing packages after installing these three packages. So let me show you how to install all unofficial library used in this project.**
## Packages that are needed to be installed:
`sudo apt-get install ros-foxy-xacro -y`
`sudo apt install ros-foxy-nav2-bringup`
`sudo apt install ros-foxy-joy-linux`
`sudo apt install ros-foxy-joy`
`sudo apt install ros-foxy-joint-state-publisher`
`sudo apt install ros-foxy-robot-localization`
## Continue with installation and build
As shown in **install_linorobot2.bash** you must do the following:

`rosdep update && rosdep install --from-path src --ignore-src -y --skip-keys microxrcedds_agent`
`colcon build`
`source $WORKSPACE/install/setup.bash`

or automatically run the script again

`bash install_linorobot2.bash <robot_type> <laser_sensor> <depth_sensor>`

### **References**
One good reference that enlighten my way:https://stackoverflow.com/questions/77899289/how-to-solve-the-error-cannot-locate-rosdep-definition-for-pcl