linorobot/linorobot2 #71: Ros2 distro : humble with linirobot2.... problem with linorobot2_hardware

hemsinghb, 2023-05-12:
As there is no ROS2 distro : humble supporting linorobot2_hardware , I tried with galactice hardware which is on github , but failed , while launching slam.launch ( ERROR : lookup would require extrapolation into the future. Requested time xxxxxx
but the latest data is at yyyyy , when looking up transform from trame to frame [map]) error is appearing and on rviz2 map robot not moving , but rotation is updating on rviz2. Please guide further. ... regards.

Sylvania2, 2023-05-30:
Installed Humble a few weeks ago, and it worked. After a reinstall last week i also see the map moving in rviz2 and the robot rotating.
Solution ?

Best regards

Sylvania2, 2023-06-02:
@hemsinghb Hi, solved it ! Read ROS2 install guide.. you have to do bee sure that Ubuntu 22.04 is upto date before installing Humble ! Reinstall Ubuntu, do a apt update, upgrade and dist-upgrade, reboot and check again before installing Humble.

Best regards
Kim

grassjelly, 2023-06-29:
thanks for sharing @Sylvania2 . Closing this for now

closed 2023-06-29 by grassjelly