linorobot/linorobot2 #73: about navigation problem

LaiVernon, 2023-06-28:
Hi there
I am attempting to build a linorobot2 on my computer, the build is successful, and I can control the mecanum wheel car by keyboard, but there are some feature I can't to execute completely it.
1. I cannot save the map, I am not sure that whether I am using simulate environment.
2. when I execute "ros2 launch linorobot2_navigation navigation.launch.py rviz:=true sim:=true" node I cannot execute navigation feature the screen is nothing happen, but the car can be moved when I control it through keyboard.
3. I found some info log on my screen
[rviz2-10] Warning: Invalid frame ID "front_left_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-10]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-10] Warning: Invalid frame ID "front_right_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-10]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-10] Warning: Invalid frame ID "rear_left_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-10]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-10] Warning: Invalid frame ID "rear_right_wheel_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-10]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
I am not sure that whether they associate above problem.

thanks for replying, first.

my environment:
ros2 version: foxy
ubuntu version: 20.04

grassjelly, 2023-06-29:
hi, may I know how are you saving the map? For navigation, remember to set the robot’s initial pose in Rviz for Nav2 to start.

LaiVernon, 2023-06-29:
Hi @grassjelly 
thank you for the replying~
1. I following the linorobot2 [README.md](https://github.com/linorobot/linorobot2#readme) to step by step to saving the map but it always timeout even if I set a very long timeout.
2. how can I set the robot's initial pose? the navigation mechanism sometime run successfully and sometime wrong... and sometime strike again when it can work normally. below is the a part of the log:
[bt_navigator-7] [INFO] [1688009576.268255609] [bt_navigator]: Begin navigating from current location to (2.89, 5.30)
[bt_navigator-7] [ERROR] [1688009576.291747406] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
[bt_navigator-7] [WARN] [1688009576.291835518] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[planner_server-5] [WARN] [1688009576.810165305] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 2.0555 Hz
[rviz2-10] Start navigation
[bt_navigator-7] [INFO] [1688009610.573643751] [bt_navigator]: Begin navigating from current location to (2.88, -0.50)
[ERROR] [bt_navigator-7]: process has died [pid 145684, exit code -11, cmd '/opt/ros/foxy/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmp7jvnlhdu -r /tf:=tf -r /tf_static:=tf_static'].

thanks for your replying again~~

grassjelly, 2023-06-29:
what dev board are you running this on?

LaiVernon, 2023-06-29:
Hi @grassjelly 
only use simulate.

LaiVernon, 2023-06-29:
Hi @grassjelly 
about the saving map, I have reinstalled the package and it had tested successful.
thanks for the helping~

LaiVernon, 2023-06-30:
Hi @grassjelly 
do you have any think about below logs, or do you need more information I provide?
thanks for the helping.

> 2. how can I set the robot's initial pose? the navigation mechanism sometime run successfully and sometime wrong... and sometime strike again when it can work normally. below is the a part of the [...]