linorobot/linorobot2 #74: [bug]  linorobot2 gazebo unable to locate command_timeout.py

atticusrussell, 2023-07-02:
I think the current path to the scripts in the gazebo launch file is not working.
(humble branch)
```
atticus@AJR-L5Pro:~/workspace/linorobot2_ws  
$ ros2 launch linorobot2_gazebo gazebo.launch.py
[INFO] [launch]: All log files can be found below /home/atticus/.ros/log/2023-07-02-10-44-33-530258-AJR-L5Pro-29352
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'command_timeout.py' not found on the libexec directory '/home/atticus/workspace/linorobot2_ws/install/linorobot2_gazebo/lib/linorobot2_gazebo' 
```

atticusrussell, 2023-07-02:
My issue was that I was using `colcon build --symlink-install`. The symbolic link was breaking things. It worked with `colcon build` after cleaning.

Probably still a bug, but less impactful.

grassjelly, 2023-08-30:
fixed this on this commit https://github.com/linorobot/linorobot2/commit/70fd3035b83b7d6fa1c52ac8f9db8575b70f8134 . This migrates the build to ament_python

grassjelly, 2023-08-30:
closing this for now. Please re-open  if the issue still persists.

closed 2023-08-30 by grassjelly