linorobot/linorobot2 #78: linorobot2 with A2M12 LIDAR problem...

hemsinghb, 2023-09-03:
Hi dear Jimeno,
I have quite happy with  belt-driven rplidar , but belt have a limited life so I tried to upgrade and purchased A2M12.
I could able to install rplidar-ros with the help of slamtec website, and I was able to run slam ( baudrate : 256000 and scan_mode: "express/standard/Sensitivity"--all 3 modes its working fine, and saved the map. But after that, while opening map / navigation part is partially ovened on rviz and continuous  messegaes coming on terminal  saying  "f
Invalid frame ID " map" passed to canTransform argument target_frame - frame does not exist...".  I tried this on galactic as well as on humble, also tried on X86 and arm64 both. 

Please help me. 
With Regards, Hem.

grassjelly, 2023-09-03:
what do you mean by partially opened? Have you initialised the pose in Rviz?

hemsinghb, 2023-09-03:
Partially, I mean only map (borders in black lines)   appearing, no cast map, no laser etc.  
I tried by by adding  following lines  (static_transform_publisher_node) in navigation.launch file , Now its working. But I don't know its a right way or wrong...

Node(
    package='tf2_ros',
    executable='static_transform_publisher',
    name='link1_broadcaster',
    arguments=['0', '0', '0', '0', '0', '0', '1', 'map', 'odom'],
    output='screen',)

pl. do needful,

hemsinghb, 2023-09-06:
Hi  Dear sir  Grossjelly,

I could produce  navigation map  by adding few lines of node in navigation.launch file as mentioned in above post [Ros2 humble Navigation ]. But I'm not getting pose particles for localization. Please advise. Thank you.