linorobot/linorobot2 #79: Can i use linorobot without depth_sensor?

fordft168, 2023-09-05:


mich1342, 2023-09-07:
Hi,

Sure you could do that. I have done it once. Just don't call the depth sensor launch and comment all depth sensor related line in both slam and nav config file

fordft168, 2023-09-09:
and if without the encoder too? cause all my encoder broke haha

mich1342, 2023-09-09:
Yes, you could do that as long as you have other source of odometry data such as visual odometry to replace the encoders

souravchandrasaha, 2024-04-24:
> Yes, you could do that as long as you have other source of odometry data such as visual odometry to replace the encoders

I'm trying to run my AMR without depth camera and its not working . Can you share your package @mich1342 Please . that would be great help.

fordft168, 2025-04-24:
ok im complete the linorobot2 without depth already but, a bit pain cause need to build every thing in pi5

closed 2025-04-24 by fordft168