linorobot/linorobot2 #85: linorobot2 bringup issue on Ubuntumate22.04.3

hemsinghb, 2023-09-19:
Dear Sir,
when I run below code its throwing error as shown below'. However there is no error while executing slam and navigation files, please help me. 

Thank you,. HEM.

rock@rock:~/linorobot2_ws  $  ros2 launch linorobot2_bringup bringup.launch.py

[INFO] [launch]: All log files can be found below /home/rock/.ros/log/2023-09-19-22-16-15-735043-rock-48883
[INFO] [launch]: Default logging verbosity is set to INFO
**[ERROR] [launch]: Caught exception in launch (see debug for traceback): <launch.substitutions.launch_configuration.LaunchConfiguration object at 0x7ffb44442290>**

RicardoSPeres, 2023-09-19:
Something similar is happening to me, only if I set LINOROBOT2_DEPTH_SENSOR to 'realsense'. I get:
`ros2 launch linorobot2_bringup bringup.launch.py
[INFO] [launch]: All log files can be found below /home/peres/.ros/log/2023-09-19-23-46-31-783276-novabot-7236
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): <launch.substitutions.launch_configuration.LaunchConfiguration object at 0xffffad6cb0d0>`

If I leave LINOROBOT2_DEPTH_SENSOR blank and use only a laser sensor with LINOROBOT2_LASER_SENSOR (not the realsense), it works fine

hemsinghb, 2023-09-20:
Dear Juan Jimeno sir,

Alternatively: 
Can we run bringup without real sense (by disabling  " realsense in .bashrc file) in one terminal and separate realsense launcher in another terminal . OR is there any way to rectify the existing bringup.launch . Please reply. THX
...---Hems.

grassjelly, 2023-09-20:
can you try setting LINOROBOT2_DEPTH_SENSOR=“” ?

hemsinghb, 2023-09-21:
SOLVED: HUMBLE DISTRO :
presently the linorobo2_ws> src>launch  etc of humble version : linorobot_bringup whole directory have some issues. so I copied it from galactic distro. 
 its working.

hemsinghb, 2023-09-21:
Actual problem is located. Its die to addition of oakd... camera in "depth.launch.py" . Problem solved by commenting few lines of code  which arre related to oakd.