linorobot/linorobot2 #86: install_linorobot2.bash fails at line 150

luapsake, 2023-09-20:
Lines 149, 155, 159 associated with OakD support have a missing space

function install_oakd{  
needs to be changed to 
function install_oakd {

grassjelly, 2023-09-20:
fixed here: https://github.com/linorobot/linorobot2/commit/735190729aa77ea8400f187572d6882d1317fb3f . Please let me know if the problem persists

hemsinghb, 2023-09-21:
Yes, the Next 2 lines also required space before the flower bracket. total 3 places.

luapsake, 2023-09-21:
> fixed here: 7351907 . Please let me know if the problem persists

Hi Juan,Linorobot2 is great!  I cannot find you on linkedin.Paul EkasSAKE RoboticsSokul Automation

luapsake, 2023-09-25:
Hi Juan,

I am using linorobot2 as a flexible foundation robot for all the robots we
build.  I'd like to setup a parallel package that has its own urdfs that
instantiate linorobot2 configurations.  We will essentially have a fleet of
different robots operating all with linorobot2 bases (but with different
configurations).

My thought is this.
mypackage/
mypackage/robots/
mypackage/robots/robot1
mypackage/robots/robot2
mypackage/robots/robot3/urdf     /launch    /meshes ...

Do you have an idea or an example that explains how you would do this?

In our implementation, we are not using any of the linobot2 microros
implementation.  We have all our own hardware and software drivers, etc.
That should not be an issue should it?

Thank you for your thoughts.

Is there somewhere else you'd like me to ask these types of questions?

Thank you,
Paul