linorobot/linorobot2 #103: Hanging when running `ros2 launch linorobot2_bringup bringup.launch.py`

MagnusMat, 2023-11-22:
Hello,
I am running linorobot2 on a Raspberry Pi.
When I run `ros2 launch linorobot2_bringup bringup.launch.py` i get hung up on `[joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...` as seen on the screenshot below.

[image]

I appear to get the same behaviour if I run `description.launch.py`. Am i not mistaken that this script should finish running eventually?
I would then run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0` afterwards.

Running `ros2 launch linorobot2_viz robot_model.launch.py` on my host machine launches RVIZ with a visualisation of the robot, so the URDF should be published.

Thanks in advance

MagnusMat, 2023-12-21:
It was a firmware issue. New firmware fixed it.

closed 2023-12-21 by MagnusMat

Saketh-Gurram, 2024-07-09:
Hi, I am still facing this issue when using ros2 humble with a teensy4.1....any idea for a fix?

MagnusMat, 2024-07-09:
Try this firmware, that made it work for us: https://github.com/P10-Exploration-and-Patrolling-Pipeline/linorobot2_hardware/blob/Polybot_config/firmware/src/firmware.cpp .