linorobot/linorobot2 #105: docker typo ?

Sylvania2, 2023-12-13:
Hi

typo in docker-compose.yaml:

- ../linorobot2_navigation/maps:/root/linorobot2_ws/src/linorobot2/linorobot2_navigation/maps:rw)

remove ) 
- ../linorobot2_navigation/maps:/root/linorobot2_ws/src/linorobot2/linorobot2_navigation/maps:rw

Best regards
Kim

grassjelly, 2023-12-13:
this is intentional to give the host access to maps within the container and vice-versa. Is it causing any issues on your setup?

Sylvania2, 2023-12-13:
Hi

While i was trying to get this docker, docker compose thing working i had and error saying something like wrong permission parameter, and the only odd thing i found was the ) character after :rw
I will try again :-)

Best regards