linorobot/linorobot2 #113: Map continuously rotating with RPLidar A1

souravchandrasaha, 2024-03-03:
Whenever I'm trying to make a indoor area map using SLAM maps seems like rotating continuously. Which giving me pretty bad map output. what to do. I'm sure a lot of people already faced the same issue. 

My system architecture is Jetson nano with RPlidar and IMU6050. I'm not using any camera .

aimethierry, 2024-04-08:
> Whenever I'm trying to make a indoor area map using SLAM maps seems like rotating continuously. Which giving me pretty bad map output. what to do. I'm sure a lot of people already faced the same issue.
> 
> My system architecture is Jetson nano with RPlidar and IMU6050. I'm not using any camera .

Did you solve the issue?

grassjelly, 2024-04-08:
does your build have a lot of vibration (for instance due to rotating lidar)? This might be due to noisy IMU.

aimethierry, 2024-04-08:
> does your build have a lot of vibration (for instance due to rotating lidar)? This might be due to noisy IMU.

Yes, it has IMU6050 and RP liadar. And how can i solve this problem?

souravchandrasaha, 2024-04-12:
how to solve this then? how to solve IMU noise issue?

aimethierry, 2024-04-12:
> how to solve this then? how to solve IMU noise issue?

Someone suggested to use  GY-85 BMP085 Sensor Modules 9 Axis Sensor Module ITG3205 + ADXL345 + HMC5883L, since it worked fine with him

tbalatka, 2024-04-17:
I'm experiencing the same issue when using SLAM. Also, the entire rviz2 display 'jiggles'. Using IMU6050.

souravchandrasaha, 2024-04-20:
> I'm experiencing the same issue when using SLAM. Also, the entire rviz2 display 'jiggles'. Using IMU6050.

Please let me know in case you solve it . It's been so many days I have been searching for solution

tbalatka, 2024-04-20:
Yes, I will.

tbalatka, 2024-04-20:
I noticed one thing I did wrong. I ran the following commands one after
the other:

-ros2 launch linorobot2_navigation slam.launch.py
-ros2 launch linorobot2_navigation slam.launch.py rviz:=true sim:=true

The above resulted in RVIZ showing two maps one after the other. It also
caused what appeared to be 'jiggling', but in reality it was two maps
displayed one after the other.

To resolve both issues, I ran the second command on its own.

I now need to fix my mapping, which is still a mess, but much better than
before.

tbalatka, 2024-04-23:
> resulted in RVIZ displaying two maps. It also caused what appeared to be
> 'jiggling', but really it was two maps quickly displayed one after the
> UDPDATE: Mapping improved bigtime when my YDLIDAR X4 was turned 180 degrees
> and had no obstructions (wires, etc) from the robot's case. I now need to
> discover how to make the map forget objects that are no longer in the
> room...any help would be most appreciated.
> Are you running it on real robot, or simulation?

Real robot running on an i3 laptop.

tbalatka, 2024-04-24:
Now I can navigate my 2wd robot in RVIZ2 and the generated SLAM map is looking more and more accurate. 
However, the TF information is not being published: I can only see the robot moving in the grid without rotation of the axis on both wheels. So I installed urdf_tutorial to test my 2wd_properties.urdf.xacro file, and indeed I can simulate left_wheel_joint and right_wheel_joint movement. Odd....

souravchandrasaha, 2024-05-06:
> > how to solve this then? how to solve IMU noise issue?
> 
> Someone suggested to use GY-85 BMP085 Sensor Modules 9 Axis Sensor Module ITG3205 + ADXL345 + HMC5883L, since it worked fine with him, However i did not test.

I'm searching for ros2 driver of this IMU since I'm connecting this IMU directly with Jetson nano via I2C . Please suggest a suitable ros2 driver for this to work.