linorobot/linorobot2 #117: Simulation issue

aimethierry, 2024-04-08:
Hey @grassjelly, 

First and foremost, thank you for your hard work. I am attempting to utilize this package but am getting warnings during the launch of gazebo, which I am using.
 Am on 
- Ubuntu 20.04
- ROS2 FOXY 

this is how i launch the bring up file 
`ros2 launch linorobot2_bringup bringup.launch.py base_serial_port:=/dev/ttyUSB0 joy:=true`

this is how launch the gazebo 
`ros2 launch linorobot2_gazebo gazebo.launch.py`

the following is my error. 

[image]

aimethierry, 2024-04-10:
> run

> Hey, you can't run gazebo.launch.py and bringup.launch.py at the same time. If you're running simulation., just run gazebo.launch.py

Hey @grassjelly , 

Now am running the package on a real robot, However, when i open a new terminal to run gazebo after the bring up i always get the same issue, Do you have an idea on how i can solve this problem if TF. 

These are the command i am using 
`ros2 launch linorobot2_bringup bringup.launch.py use_sim_time:=False`
`ros2 launch linorobot2_gazebo gazebo.launch.py`

mich1342, 2024-04-19:
In real robot you just run the bringup without the gazebo

aimethierry, 2024-04-19:
> In real robot you just run the bringup without the gazebo

Hey @mich1342 , 

Thanks for your response, by not launching gazebo i cant set initial position