linorobot/linorobot2 #118: Navigation Issue

aimethierry, 2024-04-12:
Hey @hippo5329 , @grassjelly , @madgrizzle , @mich1342 

Hey i am attempting to do navigation using; using 

1. Ubuntu 20.04
2. ROS2 FOXY

The following are the commands am running 

Bringup file 
`ros2 launch linorobot2_bringup bringup.launch.py`

Gazebo
`ros2 launch linorobot2_gazebo gazebo.launch.py`

Rviz 
`ros2 launch linorobot2_navigation navigation.launch.py map:=./src/linorobot2/linorobot2_navigation/maps/klab.yaml sim:=True rviz:=True`

**Result** 

By setting initial position on map and set nav to goal in rviz2, as a result the robot moves in simulation but it does not do anything on a real robot and  the following is TF view frame 

[image]

mich1342, 2024-04-12:
Hi, 

Please read the documentation carefully. All of your commands are for simulation only, not for real robot.

aimethierry, 2024-04-12:
> Hi,
> 
> Please read the documentation carefully. All of your commands are for simulation only, not for real robot.

Can i see the command for the real robot please, cant see them in the documentation

aimethierry, 2024-04-12:
> > Hi,
> > Please read the documentation carefully. All of your commands are for simulation only, not for real robot.
> 
> Can i see the command for the real robot please, cant see them in the documentation

@mich1342 

Can you please help on the commands to use on real robot, i spent so many hours trying but seems like am not understanding the process of doing it

aimethierry, 2024-04-12:
> @mich1342
>
> Can you please help on the commands to use on real robot, i spent so many hours trying but seems like am not understanding the process of doing it

aimethierry, 2024-04-12:
@mich1342

Can you please help on the commands to use on real robot, i spent so many hours trying but seems like am not understanding the process of doing it

closed 2024-04-12 by aimethierry

hippo5329, 2024-04-24:
1. Remove the Gazebo. It is not needed for real robot. Do not run
ros2 launch linorobot2_gazebo gazebo.launch.py

2. Remove the sim:= true in navigation command.
Rviz
ros2 launch linorobot2_navigation navigation.launch.py map:=./src/linorobot2/linorobot2_navigation/maps/klab.yaml rviz:=True