linorobot/linorobot2 #121: Using 2 ESP32 Devkit's

cantkntmr, 2024-05-21:
Hey everyone! I want to use two ESP32s for motor work because dealing with encoders is already challenging for a single ESP32. So, I plan to use one ESP32 solely for handling the encoders, and the other ESP32 for the remaining tasks. Do you think this is possible? If so, how?

hippo5329, 2024-05-24:
ESP32 has hardware pulse counters for encoders. They can count up to MHz. There is no interrupt required. There is no problem to drive up to 4 motors with encoders. Please check my wiki at https://github.com/hippo5329/linorobot2_hardware/wiki .

cantkntmr, 2024-05-24:
The problem with your wiki is, it is for Humble only. I am using Jetson
Xavier NX DEV Kit  as my robot computer. Nvidia does not provide any update
for that kit. Thus, unfortunately I am not able use your wiki.

hippo5329, 2024-05-24:
Are you using ubuntu 20.04 and foxy?

cantkntmr, 2024-05-24:
Yes, exactly.

hippo5329, 2024-05-24:
Ok. I will check pico with foxy and let you know.

cantkntmr, 2024-05-24:
That would be great. Thanks!

hippo5329, 2024-05-24:
I have a quick test on ubuntu 20.04 and foxy. There are some issues in linorobot2 installation which I think can be resolved later. However, the micro-ROS agent is built without problem. The linorobot2_hardware can not build with foxy setting. But it can be set to humble and build the firmware. And esp32 and pico firmware built as humble on ubuntu 20.04 can work with micro-ROS agent on foxy. As you will use Jetson as robot computer, I would suggest pico as the micro-controller. The USB CDC of pico will work better. And pico pin assignments are easier to use. The pico pin assignment on my wiki is tested. I have a Mecanum drive robot (4 motors with encoders) built with rpi5 and pico. Video and photo on my wiki. They work well.

cantkntmr, 2024-05-24:
Thank you but unfortunately I do not have pico. My setup is Jetson Xavier
Nx, ESP32, two L298N Motor drivers. I am very new to ROS2. Can you explain
how to build linorobot2_hardware on ESP32?

hippo5329, 2024-05-24:
ESP32 and L298N will be fine. Please follow my wiki.

hippo5329, 2024-05-24:
If you have questions, you may use my discussion board athttps://github.com/hippo5329/linorobot2_hardware/discussions .

"Issue" is mostly used for bugs reporting.
Q&A should go to "discussion".

hippo5329, 2024-05-26:
### Install linorobot2 on Ubuntu 20.04 and Foxy

Use the same procedures as humble and replace humble with foxy. However, rosdep seems failed to work. The packages dependency need to be handled manually before linorobot2 installation.

    sudo apt install -y ros-foxy-imu-tools ros-foxy-robot-localization ros-foxy-joy-linux \
        ros-foxy-xacro ros-foxy-depth-image-proc ros-foxy-joint-state-publisher \
        ros-foxy-image-proc ros-foxy-laser-filters ros-foxy-navigation2 \
        ros-foxy-nav2-bringup ros-foxy-turtlebot3-gazebo

Build the firmware with Humble setting. It will work with micro-ROS agent on Foxy. Uncomment the line of your board env in platformio.ini.

    board_microros_distro = humble

cantkntmr, 2024-05-26:
I will try it as soon as possible and let you know. Thanks!!

hippo5329, 2024-05-28:
I added more details on micro-ros agent connection test and bringup launch. Please check the wiki.