linorobot/linorobot2 #123: linorobot2 : when moves in real environment hitting objects which are left /right side which are not in same height of LIDAR plane

hemsinghb, 2024-05-30:
If anyone made a sonar range sensor with linorobot2 , please share the code to avoid obstacles.
Regars, 
Hem.

hippo5329, 2024-06-01:
https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/lib/range/range.cpp

hemsinghb, 2024-07-03:
Thank You dear [hippo5329] !. I did the ultrasound sensor data publishes and also got range sensor cone on RVIZ. 
Only issue is costmap_2d is not forming when place any obstruction in front of USS. only the cone size is increasing / decreasing.  what is next. 

my navigation.yaml is as under, please guide further:

global_costmap:
  global_costmap:
    ros__parameters:
      use_sim_time: false
      footprint_padding: 0.03
      update_frequency: 10.0
      publish_frequency: 10.0
      global_frame: map
      robot_base_frame: base_link
      robot_radius: 0.22 # radius set and used, so no footprint points
      resolution: 0.05
      plugins: ["static_layer", "obstacle_layer", "voxel_layer", "inflation_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: true
        observation_sources: scan base_scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: true
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
        base_scan:
          topic: /base/scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: true
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: true
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: true
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /camera/depth/color/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.01
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          clearing: true
          marking: true
          data_type: "PointCloud2"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: true
        enabled: true
        subscribe_to_updates: true
        transform_tolerance: 0.1
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.55
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true
        always_send_full_costmap: true

local_costmap:
  local_costmap:
    ros__parameters:
      use_sim_time: false
      update_frequency: 10.0
      publish_frequency: 10.0
      global_frame: odom
      robot_base_frame: base_link
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22 # radius set and used, so no footprint points
      #plugins: ["obstacle_layer", "voxel_layer", "inflation_layer"]
      plugins: ["obstacle_layer", "voxel_layer", "inflation_layer","sonar_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: true
        observation_sources: scan base_scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 5.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: false
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
          track_unknown_space: true
          inflate_unknown: true
        base_scan:
          topic: /base/scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: false
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: true
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: true
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /camera/depth/color/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.01
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          clearing: true
          marking: true
          data_type: "PointCloud2"
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.3
        cost_scaling_factor: 1.0
        inflate_unknown: true
        inflate_around_unknown: true
        
      sonar_layer:
        plugin: "nav2_costmap_2d::RangeSensorLayer"
        #plugin:"range_sensor_layer::RangeSensorLayer"
        frame_id: /range_link
        topic: /ultrasound_FL
        no_readings_timeout: 0.0
        clear_on_max_reading: true
        clear_threshold: 0.2
        mark_threshold: 1.0

---regards, Hem

hippo5329, 2024-07-03:
Indentation is very important in YAML. Please check your indentation. To show your code here, please fence your code blocks with three backticks (```) or three tildes (~~~) on the lines before and after the code block.

hippo5329, 2024-07-03:
mark_threshold: 0.8 , which is the default probabilty

hippo5329, 2024-07-03:
BTW, I should change the topic to range_sensor_layer default "sonar" in the firmware.

hippo5329, 2024-07-03:
I think these plugins should be in the global costmap. No need to repeat in local costmap.

hippo5329, 2024-07-04:
local_costmap: move sonar_layer before inflation_layer
global_costmap: add sonar_layer before inflation_layer

hippo5329, 2024-07-04:
```
  sonar_layer:
    plugin: "nav2_costmap_2d::RangeSensorLayer"
    topics: ["/ultrasound_FL"]
    clear_on_max_reading: true
```

hemsinghb, 2024-07-05:
hi dear Friends and Sirs.

From the above advises I just edited my .yaml file as follows:

'''global_costmap:
  global_costmap:
    ros__parameters:
      use_sim_time: false
      footprint_padding: 0.03
      update_frequency: 10.0
      publish_frequency: 10.0
      global_frame: map
      robot_base_frame: base_link
      robot_radius: 0.22 # radius set and used, so no footprint points
      resolution: 0.05
      plugins: ["static_layer", "obstacle_layer", "voxel_layer", "inflation_layer","sonar_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: true
        observation_sources: scan base_scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: true
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
        base_scan:
          topic: /base/scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: true
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: true
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: true
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /camera/depth/color/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.01
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          clearing: true
          marking: true
          data_type: "PointCloud2"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: true
        enabled: true
        subscribe_to_updates: true
        transform_tolerance: 0.1
      sonar_layer:
        plugin: "nav2_costmap_2d::RangeSensorLayer"
        enabled: True
        topics: ["ultrasound_FL"]
        no_readings_timeout: 0.0
        clear_on_max_reading: true
        clear_threshold: 0.2
        mark_threshold: 0.8
        inflate_cone: 0.99         
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.3
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true
      always_send_full_costmap: true

local_costmap:
  local_costmap:
    ros__parameters:
      use_sim_time: false
      update_frequency: 5.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22 # radius set and used, so no footprint points
      resolution: 0.05
      plugins: ["obstacle_layer", "voxel_layer", "inflation_layer","sonar_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: true
        observation_sources: scan base_scan
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        combination_method: 1
        scan:
          topic: /scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: false
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
        base_scan:
          topic: /base/scan
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          max_obstacle_height: 2.0
          min_obstacle_height: 0.0
          clearing: false
          marking: true
          data_type: "LaserScan"
          inf_is_valid: false
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
        enabled: true
        footprint_clearing_enabled: true
        max_obstacle_height: 2.0
        publish_voxel_map: true
        origin_z: 0.0
        z_resolution: 0.05
        z_voxels: 16
        max_obstacle_height: 2.0
        unknown_threshold: 15
        mark_threshold: 0
        observation_sources: pointcloud
        combination_method: 1
        pointcloud:  # no frame set, uses frame from message
          topic: /camera/depth/color/points
          max_obstacle_height: 2.0
          min_obstacle_height: 0.01
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          clearing: true
          marking: true
          data_type: "PointCloud2"
      sonar_layer:
        plugin: "nav2_costmap_2d::RangeSensorLayer"
        enabled: True
        topics: ["ultrasound_FL"]
        no_readings_timeout: 0.0
        clear_on_max_reading: true
        clear_threshold: 0.2
        mark_threshold: 0.8
        inflate_cone: 0.99           
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 0.3
        cost_scaling_factor: 1.0
        inflate_unknown: false
        inflate_around_unknown: true'''

But the costmap is not forming on rviz with USS (RangeSensorLayer). I will be grateful for your help to complete my project. 

---regards., Hem.

hippo5329, 2024-07-05:
The order of plugins is important.

global
      plugins: ["static_layer", "obstacle_layer", "voxel_layer", "sonar_layer", "inflation_layer"]

local
      plugins: ["obstacle_layer", "voxel_layer","sonar_layer", "inflation_layer"]

hemsinghb, 2024-07-05:
Dear sir, Thank you for your response .

I'm getting  this warning on terminal continuously.:

[global_costmap.global_costmap]: No range readings received for 33.50
seconds, while expected at least every 1.00 seconds.

~~~
      sonar_layer:
        plugin: "nav2_costmap_2d::RangeSensorLayer"
        enabled: True
        topics: ['ultrasound_FL']
        no_readings_timeout: 1.0       #
        clear_on_max_reading: true
        clear_threshold: 0.2
        mark_threshold: 0.8
        inflate_cone: 1.0
        clear_on_max_reading: true
        input_sensor_type: ALL
~~~

hemsinghb, 2024-07-05:
the sequence is changed, however the message continuously on terminal is :

"No range readings received for --- seconds, while expected at least every 1.00 seconds" 
as in my code "no_readings_timeout: 1.0  " is there. 

Further help please .

hippo5329, 2024-07-05:
The topic must not be relative.
  topics: ["/ultrasound_FL"]

Please revert to "no_readings_timeout: 0.0".

And check "ros2 topic hz /ultrasound_FL".

hemsinghb, 2024-07-06:
Thanks for the tip!

I did :      no_readings_timeout: 0.0

every thing is ok, no messages, but no costmap for sonar obstruction.

May I ask you please, your "sonar.launch.py"  and sonar.node  / package
file /urdf file etc. etc .

Regards, Hem

[global_costmap.global_costmap]: No range readings received for 34.22
seconds, while expected at least every 1.00 seconds

hippo5329, 2024-07-06:
What is the rate of "ros2 topic hz /ultrasound_FL" ?

hemsinghb, 2024-07-06:
Hi, Dear Thomas sir,

bringup.launch.py:
        Node(
            package='tf2_ros',
            executable='static_transform_publisher',
            output='screen',
            arguments=['0.0', '0.0', '0.0', '0', '0.0', '0.0', 'base_link',
'range_link' , '20' ]

1. range directory  which is in "linorobot2_hardware". and firmware.ino in
src/firmware/ : attached here.
2. my ping sensors are on teensy4.1
3. do you suggest to put on jetson ?
4. bringup node is as above.
5. no  sonar.urdf file in project
6. No sonar.launch.py in my project.

if you permit I will post my hardware and ws directory.

thank you.

Hem.

hemsinghb, 2024-07-06:
Sr, I tried with this also :
        Node(
            package='tf2_ros',
            executable='static_transform_publisher',
            output='screen',
            arguments=['0.0', '0.0', '0.0', '0', '0.0', '0.0', 'base_link',
'range_link']

hemsinghb, 2024-07-06:
ros2 topic hz /utrasound_FL

Result:

min: 0.020s max: 0.027s std dev: 0.00057s window: 516
average rate: 46.156
min: 0.020s max: 0.027s std dev: 0.00060s window: 563
average rate: 46.164
min: 0.020s max: 0.027s std dev: 0.00059s window: 610
average rate: 46.168
min: 0.020s max: 0.027s std dev: 0.00058s window: 657
average rate: 46.165
min: 0.019s max: 0.027s std dev: 0.00058s window: 704
average rate: 46.153
min: 0.019s max: 0.027s std dev: 0.00058s window: 751
average rate: 46.160
min: 0.019s max: 0.027s std dev: 0.00057s window: 798

hippo5329, 2024-07-06:
It might be a bug. I will look into this later. Thanks.

hemsinghb, 2024-07-07:
Hi ,

The bug is located on my .ino file .

 changed Frame_id : "/range_link"    to  Frame_id : "range_link" in  .ino hardware file, Now the costmap is working.

Thank you.