linorobot/linorobot2 #124: Build failed for esp32

iiZtZahran, 2024-06-03:
[image]
this pops after building and here is the text format of it:
KeyError: '':
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/zahran/.platformio/platforms/raspberrypi/builder/main.py", line 253:
    target_elf = env.BuildProgram()
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProjectDeps()
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 152:
    plb = env.ConfigureProjectLibBuilder()
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1202:
    project.install_dependencies()
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 987:
    if _is_builtin(spec):
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 979:
    for lb in self.env.GetLibBuilders():
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1115:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 153:
    self.process_extra_options()
  File "/home/zahran/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 301:
    self.env.SConscript(
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/zahran/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/zahran/Downloads/linorobot2_hardware-rolling/firmware/.pio/libdeps/pico/micro_ros_platformio/extra_script.py", line 158:
    build_microros()
  File "/home/zahran/Downloads/linorobot2_hardware-rolling/firmware/.pio/libdeps/pico/micro_ros_platformio/extra_script.py", line 100:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "/home/zahran/Downloads/linorobot2_hardware-rolling/firmware/.pio/libdeps/pico/micro_ros_platformio/microros_utils/library_builder.py", line 63:
    self.download_dev_environment()
  File "/home/zahran/Downloads/linorobot2_hardware-rolling/firmware/.pio/libdeps/pico/micro_ros_platformio/microros_utils/library_builder.py", line 96:
    for repo in Sources.dev_environments[self.distro]:

hippo5329, 2024-06-03:
Maybe you did not source your ros env.

    source /opt/ros/humble/setup.bash

Or

    echo "source /opt/ros/humble/setup.bash" >> ~/.bashrc
    source ~/.bashrc

Or  Uncomment the line in firmware/platformio.ini

    [env:esp32]
    ...
    board_microros_distro = humble

hippo5329, 2024-06-03:
More info on esp32 usage.

https://github.com/hippo5329/linorobot2_hardware/wiki