linorobot/linorobot2 #132: Robot is not reaching to goal position and getting structed in between.

souravchandrasaha, 2024-11-27:
robot is not reaching to given nav2_goal position and keep making different different path and eventually mission get aborted. were able to make good enough map .

hippo5329, 2024-11-27:
1. Please post videos of rviz2 and your real robot. And picture of wiring.
2. What is your robot weight, motor rating and battery rating?
3. Check your sensors and motor driving. With my test_sensors, test_motors and test_acc.

https://github.com/hippo5329/linorobot2_hardware/wiki#build-and-run-test_sensors

hippo5329, 2024-11-27:
If your robot is heavy and driving is weak, you should lower the max acc and vol.

https://github.com/hippo5329/linorobot2_hardware/wiki#velocity-smoother---optional