linorobot/linorobot2 #133: My robot turns too much

Pattapol2005, 2024-12-08:
My robot, when setting a navigation2 goal, turns more than expected. For example, when it needs to rotate 180 degrees to turn around, it rotates about 230 degrees and then turns to the right for a while before it can move straight towards the target. When it reaches the destination, the robot keeps spinning several times before stopping. Is this issue related to the PID settings in the Hardware Config? I have attached a YouTube video of the RViz window below. Thank you.

Rplidar A1M8 / Teenny 4.0 / Mecanum wheel

https://youtu.be/rKUoffeMrdY

hippo5329, 2024-12-08:
First, you should verify your RPM with test_motors and a tachometer. You should check the "stop", distance to stop. You should also run a full loaded test_acc.

https://github.com/hippo5329/linorobot2_hardware/wiki#test-the-motors-and-encoders

Next, you should verify your xxx_config.h for the wheels dimension and distance. And the urdf,

https://github.com/hippo5329/linorobot2_hardware/wiki#configure-urdf-in-linorobot2

For Mecanum driver, you will need to update the velocity smoother.

https://github.com/hippo5329/linorobot2_hardware/wiki#velocity-smoother---optional

The left-right wheels distance should be larger than that of differential drive. You should try front-right to rear-left distance.

Pattapol2005, 2025-01-05:
When I ran the command robot@Robot-ROS:~/linorobot2_hardware-master/test_motors$ pio run -e teensy40 -t upload, I encountered an issue as shown in the image.
[image]
```
robot@Robot-ROS:~/linorobot2_hardware-master/test_motors$ pio run -e teensy40 -t upload
Processing teensy40 (board: teensy40; platform: teensy; framework: arduino)
----------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (5.0.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.159.0 (1.59) 
 - tool-teensy @ 1.159.0 (1.59) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing importlib-resources with pip at PlatformIO environment
/home/robot/.platformio/penv/bin/python -m pip install importlib-resources
Requirement already satisfied: importlib-resources in /home/robot/.platformio/penv/lib/python3.8/site-packages (6.4.5)
Requirement already satisfied: zipp>=3.1.0 in /home/robot/.platformio/penv/lib/python3.8/site-packages (from importlib-resources) (3.20.2)
Installing pyyaml with pip at PlatformIO environment
/home/robot/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/robot/.platformio/penv/lib/python3.8/site-packages (6.0.2)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/robot/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/robot/.platformio/penv/lib/python3.8/site-packages (2.0.1)
Installing empy==3.3.4 with pip at PlatformIO environment
/home/robot/.platformio/penv/bin/python -m pip install empy==3.3.4
Requirement already satisfied: empy==3.3.4 in /home/robot/.platformio/penv/lib/python3.8/site-packages (3.3.4)
Configuring teensy40 with transport serial
Downloading micro-ROS dev dependencies
KeyError: 'foxy':
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/robot/.platformio/platforms/teensy/builder/main.py", line 175:
    target_elf = env.BuildProgram()
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProjectDeps()
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piobuild.py", line 152:
    plb = env.ConfigureProjectLibBuilder()
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 1202:
    project.install_dependencies()
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 987:
    if _is_builtin(spec):
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 979:
    for lb in self.env.GetLibBuilders():
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 1115:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 153:
    self.process_extra_options()
  File "/home/robot/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/piolib.py", line 301:
    self.env.SConscript(
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/robot/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
[... 1 lines cut ...]
  File "/home/robot/linorobot2_hardware-master/test_motors/.pio/libdeps/teensy40/micro_ros_platformio/extra_script.py", line 171:
    build_microros()
  File "/home/robot/linorobot2_hardware-master/test_motors/.pio/libdeps/teensy40/micro_ros_platformio/extra_script.py", line 113:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "/home/robot/linorobot2_hardware-master/test_motors/.pio/libdeps/teensy40/micro_ros_platformio/microros_utils/library_builder.py", line 63:
[... cut ...]
teensy40       FAILED    00:00:04.014
===================== 1 failed, 0 succeeded in 00:00:04.014 =====================

``` 
I have checked mecanum_properties.urdf.xacro and lino_base_config.h, and the values match the actual measurements of my robot.
[image]
[image]
[image]

I have added a velocity smoother to navigation.yaml, but the robot still oversteers as before.
[image]

Thank you for your assistance. I've been trying to resolve this issue on my own for a while, but I still haven't been able to fix it

hippo5329, 2025-01-06:
Why do you use Foxy? What is your robot computer? Foxy does not have velocity_smoother. If you use Jetson nano, you may use docker o run humble with the pr-lidar branch of hippo5329/linorobot2.

You may set ROS_DISTRO to humble or jazzzy to build the linorobot2_hardware firmware.

hippo5329, 2025-01-06:
Docker usage,

git clone -b pr-lidar https://github.com/hippo5329/linorobot2.git
cd linorobot2/docker
sudo docker compose build
sudo docker compose up -d webtop bringup navigate rviz-nav debug

sudo docker compose logs bringup

The current test uses build-in map.

You may up slam, and then
sudo docker compose up save-map

hippo5329, 2025-01-06:
https://github.com/hippo5329/linorobot2_hardware/wiki#docker

Pattapol2005, 2025-01-06:
I'm using a Lenovo mini PC. Should I install Humble? Is that correct?

hippo5329, 2025-01-06:
For mini PC, you should install ubuntu 24.04 and ros2 jazzy. They are the latest LTS version.

Pattapol2005, 2025-01-13:
Which ROS 2 distro can I use without needing Docker? At the moment, I'm using Ubuntu 22.04 with Humble, but I'm having trouble installing micro_ros_agent. Thank you.

Pattapol2005, 2025-01-13:
I'm currently using Ubuntu 22.04 with ROS2 Humble. I can run test_acc and test_motors successfully, but when I use the command:
`ros2 launch linorobot2_navigation navigation.launch.py map:=room.yaml rviz:=true`
I encounter a problem. Where do I need to make changes? Thank you.
`[component_container_isolated-1] [INFO] [1736795244.882822077] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-2] [INFO] [1736795244.959280369] [rviz2]: Message Filter dropping message: frame 'laser' at time 1736795243.292 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1736795245.119261736] [rviz2]: Message Filter dropping message: frame 'laser' at time 1736795243.439 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1736795245.151590569] [rviz2]: Message Filter dropping message: frame 'odom' at time 1736795243.136 for reason 'discarding message because the queue is full'`

hippo5329, 2025-01-13:
Please post the result from test_motors and test_acc.

Please post the message of bringup. Please check topic hz of /imu/data, /scan.

I updated my wiki. Please skip linorobot2's nav2 instructions. And follow the official nav2 guide.

https://github.com/hippo5329/linorobot2_hardware/wiki#custom-nav2_params-file

Pattapol2005, 2025-01-13:
**Result from test_acc.**
```
MIN ACC  -0.26  -0.26 m/s2   -1.42 rad/s2
IMU ACC  15.48 -14.14 m/s2
time to 0.9x max vel   0.18 sec
distance to stop   0.00 rad
BAT  0.00V MIN  0.00V
MAX PWM 32767.5 -32767.5
MAX VEL   0.01   0.01 m/s    0.03 rad/s
MIN VEL  -0.01  -0.01 m/s   -0.03 rad/s
MAX ACC   0.26   0.26 m/s2    1.42 rad/s2
MIN ACC  -0.26  -0.26 m/s2   -1.42 rad/s2
IMU ACC  14.91 -12.11 m/s2
time to 0.9x max vel   0.12 sec
distance to stop   0.00 m
BAT  0.00V MIN  0.00V
MAX PWM 32767.5 -32767.5
MAX VEL   0.01   0.01 m/s    0.03 rad/s
MIN VEL  -0.01   0.00 m/s    0.00 rad/s
MAX ACC   0.26   0.26 m/s2    1.42 rad/s2
MIN ACC  -0.26  -0.26 m/s2   -1.42 rad/s2
IMU ACC   9.85 -10.73 m/s2
time to 0.9x max vel   0.52 sec
distance to stop   0.00 rad
BAT  0.00V MIN  0.00V
MAX PWM 16383.8 -16383.8
MAX VEL   0.00   0.00 m/s    0.00 rad/s
MIN VEL   0.00   0.00 m/s    0.00 rad/s
MAX ACC   0.00   0.00 m/s2    0.00 rad/s2
MIN ACC   0.00   0.00 m/s2    0.00 rad/s2
IMU ACC   6.97  -5.13 m/s2
time to 0.9x max vel   1.00 sec
distance to stop   0.00 m
BAT  0.00V MIN  0.00V
MAX PWM 16383.8 -16383.8
MAX VEL   0.00   0.00 m/s    0.00 rad/s
MIN VEL   0.00   0.00 m/s    0.00 rad/s
MAX ACC   0.00   0.00 m/s2    0.00 rad/s2
MIN ACC   0.00   0.00 m/s2    0.00 rad/s2
IMU ACC   6.86  -3.95 m/s2
time to 0.9x max vel   1.00 sec
distance to stop   0.00 rad
BAT  0.00V MIN  0.00V
MAX PWM 16383.8 -16383.8
MAX VEL   0.00   0.00 m/s    0.00 rad/s
MIN VEL   0.00   0.00 m/s    0.00 rad/s
MAX ACC   0.00   0.00 m/s2    0.00 rad/s2
MIN ACC   0.00   0.00 m/s2    0.00 rad/s2
IMU ACC   5.63  -4.83 m/s2
time to 0.9x max vel   1.00 sec
distance to stop   0.00 m
BAT  0.00V MIN  0.00V
MAX PWM 16383.8 -16383.8
MAX VEL   0.00   0.00 m/s    0.00 rad/s
MIN VEL   0.00   0.00 m/s    0.00 rad/s
MAX ACC   0.00   0.00 m/s2    0.00 rad/s2
MIN ACC   0.00   0.00 m/s2    0.00 rad/s2
IMU ACC   3.79  -3.41 m/s2
time to 0.9x max vel   1.00 sec
distance to stop   0.00 rad
BAT  0.00V MIN  0.00V
``` 

**Result from test_motors.**
```
MOTOR1 FWD RPM    219.7     -1.4      0.1     -6.4
MOTOR1 FWD RPM    228.7     -1.4      0.0     -0.4
MOTOR1 FWD RPM    228.9     -0.9      0.1     -1.0
MOTOR1 FWD RPM    229.1     -1.1      0.2     -0.8
MOTOR1 FWD RPM    229.0     -1.0      0.3     -0.6
[... 108 lines cut ...]
[joint_state_publisher-3] [INFO] [1736796647.755214850] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[rplidar_composition-5] [INFO] [1736796648.159698550] [rplidar_composition]: current scan mode: Sensitivity, sample rate: 8 Khz, max_distance: 12.0 m, scan frequency:10.0 Hz
``` 

 **Hz of /imu/data and /scan.**
```
robot-ros@Robot-ROS:~/ros2_ws$ ros2 topic hz /imu/data
average rate: 49.964
	min: 0.015s max: 0.025s std dev: 0.00233s window: 51
average rate: 50.010
	min: 0.015s max: 0.026s std dev: 0.00200s window: 102
average rate: 49.989
	min: 0.015s max: 0.026s std dev: 0.00170s window: 152
average rate: 49.993
	min: 0.015s max: 0.026s std dev: 0.00164s window: 203
^Crobot-ros@Robot-ROS:~/ros2_ws$ ros2 topic hz /scan
average rate: 6.709
	min: 0.144s max: 0.159s std dev: 0.00408s window: 8
average rate: 6.708
	min: 0.144s max: 0.161s std dev: 0.00439s window: 15
average rate: 6.730
	min: 0.144s max: 0.161s std dev: 0.00370s window: 22
average rate: 6.724
	min: 0.144s max: 0.161s std dev: 0.00388s window: 29
```

hippo5329, 2025-01-13:
Your robot xxx_config.h might be wrong. It should be,

#define PWM_BITS 10                         // PWM Resolution of the microcontroller
#define PWM_FREQUENCY 20000                 // PWM Frequency

Then run test_acc again.

Pattapol2005, 2025-01-14:
I have changed the PWM to 10 bits and set the frequency to 20 kHz. 
These are the results of test_acc.

```
MAX PWM 1023.0 -1023.0
MAX VEL   1.46   0.02 m/s    0.20 rad/s
MIN VEL  -1.45  -0.03 m/s   -0.23 rad/s
MAX ACC  29.41   0.65 m/s2    6.73 rad/s2
MIN ACC -29.29  -1.03 m/s2   -8.14 rad/s2
IMU ACC  16.71 -14.41 m/s2
time to 0.9x max vel   0.52 sec
distance to stop   0.07 m
BAT  0.00V MIN  0.00V
MAX PWM  511.5 -511.5
MAX VEL   0.72   0.03 m/s    0.13 rad/s
MIN VEL  -0.71  -0.01 m/s   -0.13 rad/s
MAX ACC  14.55   0.84 m/s2    5.32 rad/s2
MIN ACC -14.80  -0.78 m/s2   -4.61 rad/s2
IMU ACC   9.66 -10.50 m/s2
time to 0.9x max vel   0.24 sec
distance to stop   0.03 m
BAT  0.00V MIN  0.00V
MAX PWM  511.5 -511.5
MAX VEL   0.03   0.02 m/s    3.90 rad/s
MIN VEL  -0.03  -0.01 m/s   -3.88 rad/s
MAX ACC   0.78   0.58 m/s2   81.47 rad/s2
MIN ACC  -0.91  -0.52 m/s2  -81.12 rad/s2
IMU ACC  10.35  -7.28 m/s2
time to 0.9x max vel   0.14 sec
distance to stop   0.16 rad
BAT  0.00V MIN  0.00V
MAX PWM  255.8 -255.8
MAX VEL   0.33   0.03 m/s    0.19 rad/s
MIN VEL  -0.32  -0.02 m/s   -0.06 rad/s
MAX ACC   6.79   0.71 m/s2    5.67 rad/s2
MIN ACC  -6.53  -0.58 m/s2   -3.54 rad/s2
IMU ACC   3.91  -4.56 m/s2
time to 0.9x max vel   0.28 sec
distance to stop   0.01 m
BAT  0.00V MIN  0.00V
MAX PWM  255.8 -255.8
MAX VEL   0.01   0.02 m/s    1.78 rad/s
MIN VEL  -0.02  -0.02 m/s   -1.74 rad/s
MAX ACC   0.71   0.52 m/s2   35.07 rad/s2
MIN ACC  -0.58  -0.52 m/s2  -36.84 rad/s2
IMU ACC   3.37  -3.30 m/s2
time to 0.9x max vel   0.16 sec
distance to stop   0.07 rad
BAT  0.00V MIN  0.00V
MAX PWM  255.8 -255.8
MAX VEL   0.33   0.02 m/s    0.13 rad/s
MIN VEL  -0.32  -0.01 m/s   -0.05 rad/s
MAX ACC   6.72   0.97 m/s2    4.25 rad/s2
MIN ACC  -6.72  -0.58 m/s2   -3.90 rad/s2
IMU ACC   3.83  -5.06 m/s2
time to 0.9x max vel   0.28 sec
distance to stop   0.01 m
BAT  0.00V MIN  0.00V
MAX PWM  255.8 -255.8
MAX VEL   0.02   0.03 m/s    1.79 rad/s
MIN VEL  -0.02  -0.01 m/s   -1.74 rad/s
MAX ACC   0.65   1.16 m/s2   34.72 rad/s2
MIN ACC  -0.65  -0.78 m/s2  -36.84 rad/s2
IMU ACC   2.87  -2.91 m/s2
time to 0.9x max vel   0.14 sec
distance to stop   0.07 rad
BAT  0.00V MIN  0.00V

```

hippo5329, 2025-01-14:
The test_acc is fine this time. Your robot is heavy. Please bringup and proceed to nav2 slam with default parameters as I said yesterday.

First move forward and backward. Do not turn. Check the error between laser scan and map, to see the linear localization error.

Then turn only, to see the rotation error.

Next, try mecanum default and reduced acceleration with custom nav2_params.

'''
velocity_smoother:
ros__parameters:
    ...

# default mecanum

    max_velocity: [0.5, 0.5, 2.5]
    min_velocity: [-0.5, -0.5, -2.5]
    max_accel: [2.5, 2.5, 3.2]
    max_decel: [-2.5, -2.5, -3.2]

# reduced acc

    max_velocity: [0.5, 0.5, 2.5]
    min_velocity: [-0.5, -0.5, -2.5]
    max_accel: [1.2, 1.2, 3.2]
    max_decel: [-1.2, -1.2, -3.2]

```

The power requirement of A1 lidar is strict. If your power support to Pi 4 is adjustable, set the voltage to 5.2V.

Pattapol2005, 2025-01-14:
The SLAM is working well. When the robot moves forward, backward, and rotates left and right, the laser scan aligns correctly with the robot's actual orientation.

https://github.com/user-attachments/assets/62908cb7-5f0a-4c20-b1ad-e9ea9a119a2a

But when running nav2_bringup, a problem occurs.
```
ros2 launch nav2_bringup navigation_launch.py
``` 
```
[planner_server-3] [INFO] [1736887274.074901806] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
``` 
[image]

hippo5329, 2025-01-14:
There is some minor rotation error.

The sequence of lauch is
1. bringup
2. nav2_bringup navigation_launch.py
3. slam
4. rviz2

So before slam gives map, navigation need to wait for map. There will be some error messages before nav get map from slam.

Pattapol2005, 2025-01-15:
Am I doing it correctly? I’ve tried to run Autonomous Navigation by myself, but I haven’t succeeded yet. Could you please tell me how to run Autonomous Navigation? Thank you.

https://github.com/user-attachments/assets/8c5d8e9a-c032-4c1a-b857-1e592c7a4ae8

hippo5329, 2025-01-15:
Look great.

What do you mean by "Autonomous Navigation"?

Pattapol2005, 2025-01-15:
I mean, after I've performed SLAM and obtained the map as a .yaml file, how do I load and use that map?

hippo5329, 2025-01-15:
I reverted to old linorobot2 instructions. Please follow the updated wiki,

https://github.com/hippo5329/linorobot2_hardware/wiki#slam

hippo5329, 2025-01-16:
Navigation with a map

https://github.com/hippo5329/linorobot2_hardware/wiki#nav2-navigate-with-a-map

Pattapol2005, 2025-01-18:
[image]
I set a goal for the robot, and it moves close to the target but cannot actually reach it. Which parameter should I adjust? Previously, I set the goal for it to move to different rooms, but when it reached the doorway, the robot couldn’t pass through—even though in real life, it would have plenty of space on both sides. I tried adjusting the inflation_radius to 0.3 and the cost_scaling_factor to 2.0, and the robot was able to pass through the doorway. However, now the problem is that the robot moves near the goal but can’t actually reach it. What parameter should I change? Thank you.

Pattapol2005, 2025-02-09:
I encountered an issue where the robot reaches its destination and attempts to align to the set angle, but it seems unable to achieve the desired orientation. I adjusted the yaw_goal_tolerance to a lower value, but when set below 0.1, the robot triggers multiple recoveries. According to the video, I want the robot's wheels to be parallel to the line I placed. How can I resolve this issue? Thank you!
https://github.com/user-attachments/assets/9303dc8b-7168-47ca-ae67-303635c18c74

hippo5329, 2025-02-09:
You may run test_motor and check the rpm with a laser tachometer. You may run test_acc to find the max acc/vel. Then adjust velocity smoother.

https://github.com/hippo5329/linorobot2_hardware/wiki#test-the-motors-and-sensors

https://github.com/hippo5329/linorobot2_hardware/wiki#test-the-robot-at-maximum-velocity-and-acceleration-with-full-load

https://github.com/hippo5329/linorobot2_hardware/wiki#velocity-smoother

hippo5329, 2025-02-10:
I see minor rotation error in your previous tests. We might need to fine tune rotation error. What is your LR_WHEELS_DISTANCE in linorobot2_hardware config file? For Mecanum drive, this will be different from 2wd. Try increase this value and check rotation error.

Do you run ROS2 Jazzy? What is your velocity smoother parameters?

Pattapol2005, 2025-04-02:
Hello, I have created another robot, which is a delivery robot. I placed the cargo area on top of the robot, and the Lidar is located in the center of the robot. Around the Lidar, there are posts for placing the cargo. This setup causes the Lidar to detect the posts. I found a file in the bringup config called "box_laser_filter.yaml," so I tried enabling it, but when I run the "navigation.launch.py" file, the Lidar still detects its own posts. Could you please suggest a solution to fix this? Thank you.