linorobot/linorobot2 #134: could not choose the laser sensor.

retrodorian, 2024-12-26:
Hi, when I use the command "bash install_linorobot2.bash <robot_type> <laser_sensor> <depth_sensor>" as bash "install_linorobot2.bash 4wd" a1, (I do not have depth_sensor yet so I am not choosing any) the terminal output says that I should use rplidar etc. How the system will know the model of my LIDAR if I can only specify rplidar? Also foxy has lots of rosdep issues, do you advise me to keep use it? I am using rpi4 with 4GB RAM so I have some resource issues. Thank you for your interest.

hippo5329, 2024-12-27:
BTW, you can not execute commands with cut-and-paste in ubuntu 20.04 terminal. You will need to save the commands as a script and run the script. Otherwise, some commands will not be executed.

retrodorian, 2024-12-30:
Thank you so much. I installed all dependencies manually.

closed 2024-12-30 by retrodorian