linorobot/linorobot2 #135: Lidar doesn't show

2vin2vin, 2025-01-06:
Ubuntu-22
ros2-Humble
I have launched the robot with 4wd
`ros2 launch linorobot2_gazebo gazebo.launch.py`
I have opened rviz2 and selected the LaserScan
I am unable to see any laser data
and echo of /scan gives -inf
I have been trying to understand why am I not able to see the lidar data because of which I am unable to work with slam
All of this only in simulation

Thanks

hippo5329, 2025-01-06:
If you have a real robot, you should skip simulation and proceed to real slam. If you do not have a robot, you should follow the nav2 tutorial for slam simulation on tb3/tb4.

Mohammad-Mp, 2025-01-31:
@2vin2vin : Could you fix the problem?

Prit44421, 2025-11-07:
has anyone found the solution i am having the same issue

PaulBouchier, 2025-11-07:
@Prit44421 you should not be using Humble for simulation with linorobot2. I'm closing this issue for that reason.

TL;DR There is a branch of linorobot2 called gz-sim in which simulation works with humble. But the bigger problem is there were version dependencies between gazebo and ROS-humble which made it difficult to get working. If you want to run simulation, use ROS-jazzy and the jazzy branch of linorobot2. You can even do it with a docker container, which means you don't care about the version of OS on your computer.

closed 2025-11-07 by PaulBouchier