linorobot/linorobot2 #137: Will not build past ros_micro_agent

starrtriaxle82, 2025-01-20:
=== ./uros/micro-ROS-Agent (git) ===
Cloning into '.'...
=== ./uros/micro_ros_msgs (git) ===
Cloning into '.'...
#All required rosdeps installed successfully
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [8.04s]                     
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                             
[Processing: micro_ros_agent]                                     
--- stderr: micro_ros_agent                                          
Cloning into 'xrceagent'...
HEAD is now at 57d0862 Release v2.4.2
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /opt/ros/humble/share/fastrtps/cmake/fastrtps-dynamic-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: eProsima_atomic

  Targets not yet defined: fastrtps

Call Stack (most recent call first):
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:60 (include)
  CMakeLists.txt:37 (find_package)

gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:94: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [1min 21s, exited with code 2]

Summary: 1 package finished [1min 30s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
[ros2run]: Process exited with failure 2

hippo5329, 2025-01-21:
What is your system configuration?

I make a clean install in a ubuntu 22.04 container and does not have such issue. Please follow the wiki and try out basic micro-ros example.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

https://github.com/hippo5329/linorobot2_hardware/wiki

starrtriaxle82, 2025-01-24:
System configuration:

CyberGeek compact PC
Intel Celeron N5095A @ 2GHz x 4
Mesa Intel UHD Graphics (JSL)
16 Gig Memory 
1 TB Hard Drive
Desktop Version of Ubuntu 22.04 LTS / 64 Bit
Full Desktop Version of ROS2 Humble 

So are you saying that I should run everything in a Docker?

hippo5329, 2025-01-24:
The system configuration should be fine. You should be able build without docker.

Please comment out linorobot2_ws env from .bashrc. And follow the wiki to run basic micro-ROS example.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

starrtriaxle82, 2025-01-24:
Here is the CMakeOutput.log 
At file location:home/linorobot2_ws/build/micro_ros_agent/CMakeFiles/CMakeOutput.log

[CMakeOutput.log](https://github.com/user-attachments/files/18528872/CMakeOutput.log)

I will try to #linorobot2_ws env in the .bashrc file and run the basic micro-ROS example https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

Thank you Very much for your help!!!
I will post the result.

starrtriaxle82, 2025-01-24:
After going thru the basic micro-ROS example.

allen@allen:~/uros_ws$ colcon build --symlink-install
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [8.61s]                    
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                             
[Processing: micro_ros_agent]                                     
--- stderr: micro_ros_agent                                          
CMake Warning:
  Manually-specified variables were not used by the project:

    AMENT_CMAKE_SYMLINK_INSTALL

Cloning into 'xrceagent'...
HEAD is now at 57d0862 Release v2.4.2
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /opt/ros/humble/share/fastrtps/cmake/fastrtps-dynamic-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: eProsima_atomic

  Targets not yet defined: fastrtps

Call Stack (most recent call first):
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:60 (include)
  CMakeLists.txt:37 (find_package)

gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:94: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [1min 14s, exited with code 2]

Summary: 1 package finished [1min 24s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

hippo5329, 2025-01-24:
So the problem still. Please try building in a container.

Please post env setting in the end of .bashrc.

hippo5329, 2025-01-24:
You cannot build in WSL2. You will need to use docker.

hippo5329, 2025-01-24:
It is WSL2! WSL2 might be the problem. You may try docker.

hippo5329, 2025-01-24:
If you will mount the mini PC on the robot, you should install ubuntu server. WSL2 might have more issues.

starrtriaxle82, 2025-01-30:
Wanted to update on this issue.

Downloaded and built all the files successfully on a new compact PC, same specs and device as other that wouldn't build.
I had moved from another version of ROS2 to ROS2 Humble and was running ROS1 Noetic in a workspace on my home directory on the PC that kept error'N out during the build.

Thank you again for your help and suggestions!!!!

grassjelly, 2025-01-30:
Please close if resolved. Thank you! @starrtriaxle82

closed 2025-01-30 by starrtriaxle82