linorobot/linorobot2 #138: imu_link should not be child of odom

pitosalas, 2025-01-25:
I am working to locate where I can flip the IMU TF over so that it aligns with my robot build.  I have found (without modifying anything---I think) that the imu_link tf is a child of odom which is incorrect. You can see that here:

[image]

However tracing through the (very complicated) urdf/xacro setup) I don't see it. This is the relevant file (I believe): linorobot2/linorobot2_description/urdf/sensors/imu.urdf.xacro.

Can you see where I am misunderstanding?

```
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:macro name="imu">
    <link name="imu_link"/>
    
    <joint name="imu_to_base_link" type="fixed">
      <parent link="base_link"/>
      <child link="imu_link"/>
      <origin xyz="0 0 0" rpy="0 0 0"/>
    </joint>

    <gazebo reference="imu_link">
      <gravity>true</gravity>
      <sensor name="imu_sensor" type="imu">
      <always_on>true</always_on>
      <update_rate>20</update_rate>
      <visualize>true</visualize>
      <topic>__default_topic__</topic>
      <plugin filename="libgazebo_ros_imu_sensor.so" name="imu_plugin">
          <ros>
            <remapping>~/out:=imu/data</remapping>
          </ros>
          <topicName>imu/data</topicName>
          <bodyName>imu_link</bodyName>
          <updateRateHZ>10.0</updateRateHZ>
          <gaussianNoise>0.0</gaussianNoise>
          <xyzOffset>0 0 0</xyzOffset>
          <rpyOffset>0 0 0</rpyOffset>
          <frameName>imu_link</frameName>
          <initialOrientationAsReference>false</initialOrientationAsReference>
      </plugin>
      <pose>0 0 0 0 0 0</pose>
      </sensor>
    </gazebo>
  </xacro:macro>
</robot>
```

grassjelly, 2025-01-26:
hey @pitosalas, you found the correct file to edit. Just change the rpy accordingly in the origin of the joint (imu_to_baselink)

pitosalas, 2025-01-26:
Thanks. But what about the fact that it the parent of imu_link is odom. That's wrong too isn't it?

grassjelly, 2025-01-26:
that shouldn't be `odom` but `base_link` instead. Not sure why you're getting that as it's explicitly defined to be referenced from base_link

pitosalas, 2025-01-26:
Agreed... Here is  a more detailed issue with picturesL https://github.com/hippo5329/linorobot2_hardware/issues/52

Do you see anything?

grassjelly, 2025-01-26:
I see that you're using a Madgwick filter and I believe it is also publishing IMU's TF which by default is using odom as reference frame. Assuming the documentation is the same as ROS2 https://wiki.ros.org/imu_filter_madgwick

grassjelly, 2025-01-29:
@pitosalas Please close the issue if it's related to Madgwick's default frame . Thanks

pitosalas, 2025-01-29:
We have not nailed it down yet. Turning off Madgwick's publish tf didn't fix it. But I have an issue open on Thomas' fork. But we don't have a working solution yet.

closed 2025-01-29 by pitosalas