linorobot/linorobot2 #139: Could not managed the Navigation with stability.

retrodorian, 2025-01-26:
Hello, I am using a RPI 4 / 4GB RAM. I have Ubuntu 20.04 with Ubuntu Mate Desktop, ROS2 Foxy. I implemented all the steps with using tested components (rplidar a1m8, teensy 4.1, bts7960 motor driver, 9250 imu, 2 channel encoder). I am facing with problems at navigating.  

I am giving the initial pose from RViZ. When I give a goal position, sometimes the robot go to the pose once and I can not send the new pose, and the gib amount of time the robot can not even starts to operations. It give me these type of errors:
- [WARN] [1737702882.822527732] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future. Requested time 1737702882.821602 but the latest data is at time 1737702882.777000, when looking up transform from frame [base_footprint] to frame odom]).

-[INFO] [1737702882.8226656721 [amcl]: Setting pose (1737702882.822662): 6.022 3.020 - 0.019 
[rviz2-10] Start navigation
[rviz2-101 ][ERROR] [1737702904.319517746] [rviz2]: FollowWaypoints action server is not available. Is the initial pose set?

-controller_server-4][WARN] [1737702107.230180978] [DWBLocalPlanner]: A scoring function failed to prepare
controller_server-4]
[ERROR] [1737702107.230275551] [GoalDistCritic]: None of the points of the global plan were in the local costmap.
controller_server-4]
[WARN] [1737702107.230297884] [DWBLocalPlanner]: A scoring function
controller_server-4]
[ERROR] [1737702107.262903250] [DWBLocalPlanner]: No valid trajectories out of 419!
controller_server-4]
controller_server-4]
[ERROR] [1737702107.264442277] [controller_server]: No valid trajectories out of 419!
controller_server-4]
[WARN] [1737702107.266292866] [controller_server_rclcpP_node]: [follow_path] [ActionServer] Aborting

- Also I am getting problems about the controller server and planner server frequency that thee desired frequencies could not reached. I try to decrease them from 20.0 to 15.0 by 1.0, but it got worse now the robot can not go to the point and the localization is worse. Should I change my development card or is there something else?

hippo5329, 2025-01-29:
You may try cyclonedds.

The current ROS2 stable LTS distro is Jazzy. Foxy is EOL. Do you have a reason to use Foxy?

retrodorian, 2025-01-29:
thank you @hippo5329 for the reply! the reason behind my choice of foxy is performance issues related to raspberry. I need to use the raspberry with desktop environment (for rviz etc) and the only way that I found stable is ubuntu 20.04 server with ubuntu mate desktop on raspberry pi4 4 gb ram. anything above creates performance issues. With ubuntu 20.04, I can only use foxy. Do you have any recommendations? Should I upgrade to raspberry pi 5 with 8 gb ram or is the current raspberry is sufficient and I wrong about something.