linorobot/linorobot2 #141: Best way to adjust urdf

pitosalas, 2025-01-29:
You have a sophisticated urdf with multiple subfiles and so on.

I want to change many dimensions.

I could use the urdf argument and give it a totally new urdf directory but then I would decouple from your design.

Is that what you recommend or is there another approach?

Also do you have any tools to help edit and visualize urdf (including macros etc.?)

Thanks,

Pito

grassjelly, 2025-01-29:
try to reuse as much from the components. I'd start selecting the reusable ones from the robot xacros for example: https://github.com/linorobot/linorobot2/blob/humble/linorobot2_description/urdf/robots/2wd.urdf.xacro. 

You could also just totally replace the templetized URDF here https://github.com/linorobot/linorobot2/blob/humble/linorobot2_description/launch/description.launch.py#L28 with your own, but you'll have to ensure that the Gazebo plug-ins work if you'll be simulating the robot.

pitosalas, 2025-01-29:
So it’s ok to modify robots/2wd.urdf.xacro but prefer not to modify below that.

Is there a way do make the base be round instead of rectangular without replacing all of base.urdf.xacro?

grassjelly, 2025-01-29:
Yes, it's okay to modify that. Each component in the <robot>.urdf.xacro is modular - the sensors, mechanical parts, etc. You can change the geometry of the visual: https://github.com/linorobot/linorobot2/blob/humble/linorobot2_description/urdf/mech/base.urdf.xacro#L10 and collision: https://github.com/linorobot/linorobot2/blob/humble/linorobot2_description/urdf/mech/base.urdf.xacro#L21 . 

Now that you've mentioned, it's actually a great idea to add a circular base. Let me add that to the gz-harmonic branch. Might as well.

pitosalas, 2025-01-29:
While you are at it, you might consider making laser, depth sensor and sonar, each conditional. And you might add a camera also conditional.