linorobot/linorobot2 #142: Linorobot2 for ROS2 Jazzy Jalisco on Pi5

Alwynn03, 2025-01-31:
Hi there,
love the linorobot2 repository where i used it on a pi4 using ros2 humble. But i face installation issue on pi5 that is the package isn't supported for Ros2 Jazzy. I prefer Humble but as i have a pi5 , the pi imager only gives flash option for ubuntu 24.04 and thus i am forced to go with ros2 jazzy. Would be glad if support for ros2 jazzy would be launched. My current options are using docker with ubuntu 22.04 and install humble in that OR should i install VirtualBox with ubuntu 22.04 in it? 
Confused on which would be the better option.

grassjelly, 2025-01-31:
I would recommend using Docking using Humble branch. You can use the  docker inside the project or you can use https://github.com/grassjelly/daisy  - it comes with nice tooling for development.

Alwynn03, 2025-01-31:
Thank you! Amazed by your quick response! I am a noobie on the docker concept, so could you explain how to use the docker in the project? I checked out daisy and understood that i can setup humble but how do i proceed with linorobot package?

hippo5329, 2025-02-12:
You may use Jazzy on Pi 5. The RMW on Jazzy is improved a lot.

https://github.com/hippo5329/linorobot2_hardware/wiki

You may install linorobot2 with

```
cd ~
mkdir -p linorobot2_ws/src
cd linorobot2_ws/src
git clone https://github.com/Slamtec/sllidar_ros2.git
git clone https://github.com/linorobot/linorobot2.git
cd linorobot2
touch linorobot2_gazebo/COLCON_IGNORE
git remote add hippo https://github.com/hippo5329/linorobot2.git
git fetch hippo
cd ../..
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install
```

Then add the env to .bashrc

```
export LINOROBOT2_BASE=2wd
export LINOROBOT2_LASER_SENSOR=a1
export LINOROBOT2_DEPTH_SENSOR=
source $HOME/linorobot2_ws/install/setup.bash
```

This assume micro-ros agent installed in ~/uros_ws as in my wiki. You may later check out my branch to test.