linorobot/linorobot2 #148: Current structure of URDF is not helpful if I want to surgically customize it

pitosalas, 2025-03-06:
There are lots of hardcoded paths so that I can't just make a complete copy of the linorobot2_description/urdf it in my own package. So I am forced to modify the originals which makes it more difficult for me to incorporate your upstream improvements. This is unfortunate.

grassjelly, 2025-03-09:
I'm afraid this is as customizable as it gets already. Most of the available URDFs out there are very specific to a robot

pitosalas, 2025-03-09:
Yes I was thinking too on what I would propose. Overall the scheme works but it doesn't allow me to just make a copy of the whole urdf directory structure and modify it at will. I am forced to modify the ones that come from upstream. Which means I would want to keep the modifications to the very minimum. The more I modify the more of a merge problem I will have if/when you update yours. 

Thinking about it, the things I would want to modify in the urdf (so far) are:

- base being round rather than rectangular
- location and RPY for the IMU
- location and RPY for the LIDAR (laser)
- rendering colors for the left and right wheel (to help make clear what the front of the robot is)

I am sure there are others but that's a pretty short list.  So my proposal would be to revise the URDF to make those things also parameters in the 2wd_properties.urdf.xacro etc files.

closed 2025-03-09 by pitosalas